bohdan-s/SunGather #1: support for hybrid (sh) series of sungrow inverters

chainsawbike, 2021-12-16:
Here is the official docs for the hybrid range:

[communication-protocol-of-residential-hybrid-inverterv1.0.20-1.pdf](https://github.com/bohdan-s/SunGather/files/7724948/communication-protocol-of-residential-hybrid-inverterv1.0.20-1.pdf)

I doubt I'll have time to add them myself until late Jan, so I'll leave them here incase someone else wants to add them before then

bohdan-s, 2021-12-16:
At quick glance most of these overlap. I should be able to add these in next week. 
The model list is the key part. Update that and most the "standard" registers will work.

bohdan-s, 2021-12-16:
@chainsawbike Hi, I have added initial support for hybrid inverters.
Could you try with config options hybrid: true and level: 2 using the console export?
You should see everything up to register 13020. If it works i'll add the rest fo the registers :)

chainsawbike, 2021-12-22:
sorry, limited time right now.
detects inverter and prints 8 values, 
6 are correct
these values are 0:
export_to_grid                      
import_from_grid

ps: no battery is connected to my system ( yet... )

bohdan-s, 2021-12-24:
@chainsawbike when you get time, upload to the latest build,
It now has a http export (simple web server)
enable it in the conifg and set "level: 3" for the inverter. 
Open the url and you should see a all the registers, let me know what ones are missing / 0 and i'll look into it.

benni336, 2022-01-20:
Hi folks,
I had looked at the register.yaml and I have also noticed level 3 output.
But I think there is something wrong there.
SH10RT e.g. has 2 mptts, but on lvl 2 only mptt1 is displayed and on lvl 3 mptt1 to mptt10 are displayed.

In addition starting with level 2, I get the Error Message
```
logged 55 registers to Console
2022-01-20 15:07:14 WARNING  Modbus connection failed: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```

According to the comment in the config.yaml
```
# 1 (default) = Useful data, all required for exports, 
# 2 everything your Inverter supports, 
# 3 Everything from every register 
```
I guess here is something wrong in register.yaml matching the wrong registers to SH10RT and SH8.0RT.
That's why I wanted to review them. Are there any further known issues?

Because if not, I would say, after resolving these two mentioned one from my side, SH10 Inverters could be announced as "supported and tested"

benni336, 2022-01-20:
I would like to point out, that I added further registers in my nightly. 
(https://github.com/benni336/SunGather/tree/SH_Inverters)
They needed to be tested and so I would be happy for others to review them too.

I do not want to pull anyone away from this great project and I am happy to merge my changes back here asap.

bohdan-s, 2022-01-20:
Looks good, if you could post what model you have and PDF you are reading registers off I can upload to my other repo with the other documents here: https://github.com/bohdan-s/Sungrow-Inverter

benni336, 2022-01-20:
Find it attached.

[Communication Protocol of Residential Hybrid InverterV1.0.22_20201117.pdf](https://github.com/bohdan-s/SunGather/files/7909178/Communication.Protocol.of.Residential.Hybrid.InverterV1.0.22_20201117.pdf)

chainsawbike, 2022-01-21:
thanks for your work :)
would it be easy to show the register number alongside the name in the web view? it would make debugging a lot faster for me

bohdan-s, 2022-01-21:
Done. Will be in the next release :)

benni336, 2022-01-21:
@chainsawbike : Do you have a battery? Can you please give me some feedback on the values presented there? I don't get any values from the battery. But it might be a problem in the setup of the inverter. I will contact my eletrician for that next week to make sure it is not.

benni336, 2022-01-26:
I filed a pull request which adds a bunch of SH-Inverter variables. But there is still some todo.

One issue is the start-stop stuff. It is not yet correctly interpreted and it is more complex than for the other inverters. Additionally the status names changed. So e.g. "running" does not reflect the status correctly for SH-inverters.
I will bother with that stuff the next days.
Would be great to hear some feedback.

chainsawbike, 2022-02-14:
@benni336 no battery yet

it is producing the correct data for the values i would consider important for non battery use so i would say the hybrid series of inverters have working initial support, i am now using this to put data into home-assistant :)

bohdan-s, 2022-03-03:
see: https://github.com/bohdan-s/SungrowModbusWebClient/issues/2
Should be supported now

bohdan-s, 2022-03-07:
v0.3.6 now should have full support :) registers taken from the current documentation

closed 2022-03-07 by bohdan-s