bohdan-s/SunGather #2: Please make sure mqtt.py exists in the exports folder

rark-ha, 2021-12-17:
Upon loading I get the error message "Please make sure mqtt.py exists in the exports folder"

I can confirm it exists. I've set up the config file with the correct IP for the MQTT server, unhashed the port (but use the default 1883 anyway), and added a username/password using " " to capture them.

rvdblink, 2021-12-17:
pip3 install paho-mqtt should do the trick.

rark-ha, 2021-12-17:
Says it's already installed ("defaulting to user installation as package already installed...").

Looking at the console output I have the following error;

`Exception in thread Thread-1121:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/rarking/SunGather/SunGather/exports/mqtt.py", line 38, in publish
    self.sensor_topic = self.sensor_topic.replace('{model}', inverter.get('device_type_code', 'unknown').replace('.',''))
AttributeError: 'NoneType' object has no attribute 'replace'`

bohdan-s, 2021-12-17:
Looks like model detection is failing (as well as the other issue). Can you try adding the model to config.py and let me know if it works?
I'll add a better warning message and fail safe shortly.

rark-ha, 2021-12-17:
> Looks like model detection is failing (as well as the other issue). Can you try adding the model to config.py and let me know if it works?

Added "SG10RT" to the config.py and still get the same timeout error when loading.

bohdan-s, 2021-12-20:
> pip3 install paho-mqtt

Thanks, I've updated the requirements.txt and added it to the readme :)

bohdan-s, 2021-12-20:
@rark-ha Could you please try again (after git pull) with the following settings?
connection: sungrow
model: "SG10RT"
logging: 10

Then paste the output for me? I have updated the model detection as it didn't seem to be working correctly.

rark-ha, 2021-12-20:
Still fails for me an exits the script.

> 2021-12-21 06:33:18 INFO     Starting SunGather 0.1.0
2021-12-21 06:33:18 INFO     Loaded config: config.yaml
2021-12-21 06:33:19 INFO     Loaded registers: registers.yaml
2021-12-21 06:33:19 INFO     Loaded Export: exports\console
2021-12-21 06:33:19 INFO     Configured Console Logging
2021-12-21 06:33:19 INFO     Configured export: console
2021-12-21 06:33:19 INFO     Loaded Export: exports\mqtt
2021-12-21 06:33:24 ERROR    Failed loading export: timed out
			     Please make sure mqtt.py exists in the exports folder
2021-12-21 06:33:24 INFO     Loaded Export: exports\pvoutput
2021-12-21 06:33:24 INFO     Configured PVOutput Client
2021-12-21 06:33:24 INFO     Configured export: pvoutput
2021-12-21 06:33:24 INFO     Connection: ModbusTcpClient(192.168.20.43:502)
2021-12-21 06:33:24 DEBUG    Connection to Modbus server established. Socket ('192.168.20.244', 54219)
Traceback (most recent call last):
  File "sungather.py", line 73, in <module>
    client.connect()
  File "/home/rarking/.local/lib/python3.6/site-packages/SungrowModbusTcpClient/SungrowModbusTcpClient.py", line 54, in connect
    self._getkey()
  File "/home/rarking/.local/lib/python3.6/site-packages/SungrowModbusTcpClient/SungrowModbusTcpClient.py", line 40, in _getkey
    self._key_packet = self._recv(25)
  File "/home/rarking/.local/lib/python3.6/site-packages/pymodbus/client/sync.py", line 295, in _recv
    recv_data = self.socket.recv(recv_size)
ConnectionResetError: [Errno 104] Connection reset by peer

bohdan-s, 2021-12-20:
@rark-ha can you confirm you have the IP's correct in the config.py?
It seems the script couldnt connect to either the MQTT or Inverter IP address.

rark-ha, 2021-12-21:
> @rark-ha can you confirm you have the IP's correct in the config.py?
> It seems the script couldnt connect to either the MQTT or Inverter IP address.

!!!! it was my MQTT docker blocking IP requests. Sorry for the hassle. I had to remove the inveter model (let it auto-detect) otherwise it caused the other error above and would not publish any MQTT topics.

Topics now publishing and the inverter was added to Home Assistant with auto-discovery. But when I restarted my Home Assistant they all now show as 'Unavailable' (a second HA instance that hasn't been restarted shows the data still flowing through). Is it an availability_topic issue or other? 

I can't see any config setting to assist here?

bohdan-s, 2021-12-21:
Thats interesting about HA, I am having the same issue and thought it was a "me" issue.
I'll have a look into it.

bohdan-s, 2021-12-21:
@rark-ha I have just pushed a new commit that fixes the HA auto-discovery. Needed to set the retain flag for MQTT messages.

rark-ha, 2021-12-21:
Pulled the latest copy and that fixed it after a couple of restarts to test. Thank you @bohdan-s 

I'd be keen on the Docker version to house this great setup.

bohdan-s, 2021-12-21:
Ok, initial docker build up, still some cleaning to do but should all be working
```
docker pull bohdans/sungather
docker run -v {path to}/config.yaml:/usr/src/sungather/config.yaml --name sungather bohdans/sungather
```

rark-ha, 2021-12-21:
Excellent. Docker now running with the same config.yaml file I had setup before.

After it was running, I opened up the browser (from a different machine) to view the Sungrow local HTTP page. I then started getting this warning message (for all the register ranges) and had to restart the docker container to fix.

`WARNING  No data. Try increasing the timeout or scan interval for read, 5000:100: 'result_data' | stdout`

I then noticed that after restarting I wasn't getting any updates to PVoutput.org (though the first 10 minutes of the docker container did send it through). Nothing in the logging DEBUG suggested any error. I've got logging set to INFO now and seeing "Published to PVOutput" but getting no new data show up there.

bohdan-s, 2021-12-22:
PVOutput is a quick fix, send the timezone env to docker (-e TZ=Australia/Sydney) I have updated the readme to reflect that.
I will update the pvoutput export to use the the inverter time instead of local time, that will resolve the issue properly.

As for the other error, I can reproduce it myself. I'll have to update the SungrowModbusWebClient package and bump the required version. It seems the dongle can only have a single web session, so I will need to detect and re-login again.

rark-ha, 2021-12-22:
> PVOutput is a quick fix, send the timezone env to docker (-e TZ=Australia/Sydney)
Fixed thank you. I'd noticed a few minutes before the hour that the log started giving "Exceeding 60" errors so I knew the data must have been getting through but couldn't see anything on the website to help diagnose.

I moved the IP address to the ethernet connection successfully, and now just use the WiFi dongle to view the web session. This doesn't cause any error with a single web session for me.

@bohdan-s thanks very much for your quick responses and work on getting this git up. Solved much frustration for me.

I'll close this issue now and might open another to assist with Home Assistant Energy (as to set this up it seems to want a seperate import from grid and export to grid sensor, not a combined net sensor that goes negative).

closed 2021-12-22 by rark-ha

bohdan-s, 2021-12-22:
@rark-ha 
There are 2 computed values export_to_grid and import_from_grid that are not listed in the registers.
If you see the readme, there is a section for HA, add the sensors listed, wait ~5mins for it to generate data and then you should be able to pick those sensors for Home Assistant Energy. 
If they are returning 0 values, try enabling 
smart_meter: True 
in the config, as you need to have a smart meter to be able to read house consumption which HA Energy also reads.

arfrater, 2022-03-13:
I'm  getting this error as well new install and checked MQTT
2022-03-13 12:07:10 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
			     Please make sure mqtt.py exists in the exports folder