bohdan-s/SunGather #4: Home Assistant sensor setup and Energy platform

rark-ha, 2021-12-22:
Following on from the comment in the closed item https://github.com/bohdan-s/SunGather/issues/2#issuecomment-999236995

@bohdan-s the updates you've made in the past 48 hrs have fixed my initial issue as HA now generates 6 sensors. For others that would like a walk-through doc please read below.

In the SunGather config.yaml you need to set the smart_meter if you have one;
```
smart_meter: True
```

HA sensors created by the MQTT auto-discovery;

```
sensor.inverter_active_power
sensor.inverter_daily_generation
sensor.inverter_export_to_grid
sensor.inverter_import_from_grid
sensor.inverter_load_power
sensor.inverter_meter_power
```

Used the README settings for HA sensors and put the following into your sensors.yaml

```
sensor:
  - platform: integration
    source: sensor.inverter_active_power
  - platform: integration
    source: sensor.inverter_export_to_grid
  - platform: integration
    source: sensor.inverter_import_from_grid
```

The additional sensors then setup in HA are the following with Wh unit of measurement for the Energy platform.

```
sensor.sensor_inverter_active_power_integral
sensor.sensor_inverter_export_to_grid_integral
sensor.sensor_inverter_import_from_grid_integral
```

You need to utilise the customize.yaml manual method to get some friendly names and icons. Add the following;

```
sensor.sensor_inverter_active_power_integral:
  icon: mdi:solar-power
  friendly_name: Solar Generation
sensor.sensor_inverter_export_to_grid_integral:
  icon: mdi:transmission-tower-export
  friendly_name: Grid Export
sensor.sensor_inverter_import_from_grid_integral:
  icon: mdi:transmission-tower-import
  friendly_name: Grid Import
```

Navigate to the Energy platform (Configuration > Energy). Add these sensors to the following areas:

```
Electricity Grid > Grid Import - sensor.sensor_inverter_import_from_grid_integral
Electricity Grid > Grid Export - sensor.sensor_inverter_export_to_grid_integral
Solar Panels > Solar Production - sensor.sensor_inverter_active_power_integral
```

rark-ha, 2021-12-22:
A tip for those using PVOutput.org.

If you're uploading from SunGather using the 60 limitation (free account), and you're also trying to use the Home Assistant integration to download from PVOutput.org then you need to step up to the donation account otherwise the up/down calls will max your 60 limit out within 30 minutes of each hour.

closed 2021-12-22 by rark-ha

bohdan-s, 2021-12-24:
@rark-ha can you check the PVOutput again with the latest build (0.1.1)?
It should now respect the rate limit and spread it out over the 1 hour now. 
Make sure "rate_limit: 60" is set in the pvoutput config section.

rark-ha, 2021-12-24:
> @rark-ha can you check the PVOutput again with the latest build (0.1.1)?
> It should now respect the rate limit and spread it out over the 1 hour now.
> Make sure "rate_limit: 60" is set in the pvoutput config section.

Pulled latest and ran without issue. PVOutput recevied the figures throughout the hour without issue. I think my original issue was pushing to PVOutput with SunGather and pulling from PVOutput with Home Assistant (and hitting the 60 per hour issue around the 40min mark).

`work_state_1                             Derating Run`

It would be good to create a sensor in HA to show when it's limiting the PV production. I can only export 5KW so get derated during the peak of the day. I'd use the sensor to trigger turning on the pool pump again etc as I'm overproducing power.

bohdan-s, 2021-12-24:
Yup, should be able to add that no problem. 
I also wanted to add “is_running” but currently I have hard coded the discovery as sensors, so need to update for that as well.