bohdan-s/SunGather #6: pvoutput values don't look right

gservat, 2021-12-23:
Thanks for the rewrite @bohdan-s! I just got around to trying it today. My config is as follows:

```
inverter:
  host: sungrow.gateway
  port: 8082             # Default for modbus is 502, for http is 8082
  # slave: 0x01           # Default is 0x01
  # timeout: 10           # Default is 10
  # scan_interval: 30     # Default is 30
  connection: http     # options: modbus, sungrow, http
  # model: "SG7.0RT"      # This is autodected on startup, only needed if detection issues or for testing
                          # See model list here: https://github.com/bohdan-s/SunGather#supported
  smart_meter: True     # Default is False, Set to true if inverter supports reading grind / house consumption
  # use_local_time: False # Default False, Uses Inveter time, if try uses PC time when updating timestamps (e.g. PVOutput)
  # hybrid: False         # Default false, if you have a Hybrid (battery) set to True
  # logging: 30           # 10 = Debug, 20 = Info, 30 = Warnning (default), 40 = Error
  # level: 1              # 0 = Model and Solar Generation,
                          # 1 (default) = Useful data, all required for exports,
                          # 2 everything your Inverter supports,
                          # 3 Everything from every register

exports:
  - name: console   # Print Registers to console, good for debugging / troubleshooting
    enabled: True

  - name: pvoutput      # Publish Registers to PVOutput
    api: "<key>"
    sid: "<sid>"
    # 60 for regular accounts, 300 for donation accounts
    rate_limit: 60
    enabled: True
    parameters:
      - name: v1    # Energy Generation
        register: daily_power_yields    # Solar Generated Today (Energy)
        multiple: 1000
      - name: v2    # Power Generation
        register: total_active_power    # Current Generation (Power)
      - name: v4    # Power Consumption
        register: load_power            # Current Home usage (Power)
      - name: v5    # Temperature
        register: internal_temperature  # Inverter internal temperature
      - name: c1    # Cumulative Flag,
        value: 1    # If using v2/v4 set to 1
```

In `pvoutput.org`, I'm seeing all zero values except for temperature, which I guess makes sense based on the values of the registers:

[image]

```
Register                                 Value
device_type_code                         SG10RT
daily_power_yields                       54.0
total_power_yields                       3644.9
total_running_time                       1000
internal_temperature                     42.4
total_apparent_power                     0
total_active_power                       0
work_state_1                             Standby
meter_power                              538
load_power                               0
monthly_power_yields                     1217.1
start_stop                               Start
export_to_grid                           0
import_from_grid                         538
timestamp                                2021-12-23 21:01:07
```

I'll admit I still don't fully understand what each field in pvoutput means 😆 Given it's 9:30PM here, I was hoping to just see how much I'm consuming from the grid right now. Will be interesting to see the numbers tomorrow when the sun is out.

gservat, 2021-12-23:
Is it possible that total power used by the household is equal to `total_active_power` + `meter_power`?

rark-ha, 2021-12-23:
Just checking that in your actual config file you have the API and KEY set.

```
  - name: pvoutput      # Publish Registers to PVOutput
    api: "<key>"
    sid: "<sid>"
```

bohdan-s, 2021-12-23:
@gservat I have updated the Readme to add some clarity, thanks.
Basically:
total_active_power = power generated by inverter
meter_power = power reading at meter box

so if meter_power is +ve, power is being IMPORTED from the grid, if its -ve then its EXPORTING.
total_active_power + meter_power = house consumption

e.g. if total_active_power = 5000 (5kw generated now) + meter_power = -2000 (exporting 2kw) then your house consumption is 3000 or 3kW

bohdan-s, 2021-12-23:
@gservat if your load is always 0, I may have to add "extra" register that is total_active_power + meter_power.

You can also try "level: 2" or even 3 and see what other details your inverter returns.

gservat, 2021-12-24:
> Just checking that in your actual config file you have the API and KEY set.
> 
> ```
>   - name: pvoutput      # Publish Registers to PVOutput
>     api: "<key>"
>     sid: "<sid>"
> ```

yeah - i redacted those.. it's not an issue with the connection to pvoutput but more in the data itself.

gservat, 2021-12-24:
> @gservat if your load is always 0, I may have to add "extra" register that is total_active_power + meter_power.

Yeah, looks like `load_power` is always `0` for me.

> You can also try "level: 2" or even 3 and see what other details your inverter returns.

Quite a bit more! Getting this info:

```
Register                                 Value
device_type_code                         SG10RT
nominal_active_power                     10.0
output_type                              3PSL
daily_power_yields                       29.6
total_power_yields                       3674.5
total_running_time                       1008
internal_temperature                     48.1
total_apparent_power                     3968
mppt_1_voltage                           396.6
mppt_1_current                           4.6
mppt_2_voltage                           427.3
mppt_2_current                           4.5
total_dc_power                           3746
phase_a_voltage                          242.4
phase_b_voltage                          244.8
phase_c_voltage                          248.0
phase_a_current                          5.9
phase_b_current                          5.9
phase_c_current                          5.9
total_active_power                       3595
total_reactive_power                     1716
power_factor                             0.91
grid_frequency                           50.01
work_state_1                             Run
nominal_reactive_power                   5.0
array_insulation_resistance              3144
active_power_regulation_setpoint         11000
reactive_power_regulation_setpoint       0
work_state_2                             Stop
meter_power                              242
load_power                               0
monthly_power_yields                     1246.7
negative_voltage_to_the_ground           6200.4
bus_voltage                              706.1
pid_work_state                           0
pid_alarm_code                           0
start_stop                               Start
power_limitation_switch                  Enable
power_limitation_setting                 1100
power_factor_setting                     0.0
night_svg_switch                         Disable
reactive_power_adjustment_mode           Enable Q(U)
reactive_power_percentage_setting        65535
power_limitation_adjustment              11.0
reactive_power_adjustment                65535
pid_recovery                             Disable
export_to_grid                           0
import_from_grid                         242
is_running                               True
timestamp                                2021-12-24 12:51:41
Logged 50 registers to Console
```

rark-ha, 2021-12-24:
> ```
> total_active_power                       3595
> meter_power                              242
> load_power                               0
> export_to_grid                           0
> import_from_grid                         242
> ```

I've you're producing 3595W and your consumption is 242W then you should be exporting to the grid.

Are you sure you have a smart meter installed (or installed correctly)?

Does your PV, Grid and Load figures look correct in iSolarCloud?

gservat, 2021-12-24:
I think this is saying I'm producing (and consuming) 3595W but also drawing in 242W from the grid (total of 3837W being consumed).

bohdan-s, 2021-12-24:
@gservat i think you are right,
I think the load should be 3837W instead of 0. 
Do you have a smart meter installed with the inverter? DTU666 or something usually.

I can set load_power to a formula if 0, that will give the answer, but I wonder if its behind another register...

Can you try level: 3 (everything) and look for export_power and power_meter. These are both undocumented but used in Solariot

gservat, 2021-12-24:
@bohdan-s yes, smart meter is installed (not sure which but it's definitely there).

Just bumped level to 3 and I see `export_power` showing `1590` (I guess this means importing from the grid? as iSolarCloud also shows I'm importing roughly this amount) and `power_meter` is `0`. Full list of registers:

```
Logged 121 registers to Console
Register                                 Value
device_type_code                         SG10RT
nominal_active_power                     10.0
output_type                              3PSL
daily_power_yields                       30.6
total_power_yields                       3675.5
total_running_time                       1008
internal_temperature                     46.3
total_apparent_power                     2264
mppt_1_voltage                           393.7
mppt_1_current                           2.8
mppt_2_voltage                           425.3
mppt_2_current                           2.8
mppt_3_voltage                           0.0
mppt_3_current                           0.0
total_dc_power                           2292
phase_a_voltage                          239.2
phase_b_voltage                          242.4
phase_c_voltage                          244.8
phase_a_current                          3.4
phase_b_current                          3.4
phase_c_current                          3.4
total_active_power                       2156
total_reactive_power                     685
power_factor                             0.96
grid_frequency                           50.02
work_state_1                             Run
alarm_time_year                          0
alarm_time_month                         0
alarm_time_day                           0
alarm_time_hour                          0
alarm_time_minute                        0
alarm_time_second                        0
alarm_code_1                             0
nominal_reactive_power                   5.0
array_insulation_resistance              3144
active_power_regulation_setpoint         11000
reactive_power_regulation_setpoint       0
work_state_2                             Stop
meter_power                              1617
meter_a_phase_power                      0
meter_b_phase_power                      0
meter_c_phase_power                      0
load_power                               0
daily_export_energy                      65535
total_export_energy                      65535
daily_import_energy                      65535
total_import_energy                      65535
daily_direct_energy_consumption          65535
total_direct_energy_consumption          65535
mppt_4_voltage                           0.0
mppt_4_current                           0.0
mppt_5_voltage                           0.0
mppt_5_current                           0.0
mppt_6_voltage                           0.0
mppt_6_current                           0.0
mppt_7_voltage                           0.0
mppt_7_current                           0.0
mppt_8_voltage                           0.0
mppt_8_current                           0.0
monthly_power_yields                     1247.7
mppt_9_voltage                           0.0
mppt_9_current                           0.0
mppt_10_voltage                          0.0
[... 282 lines cut ...]
power_limitation_switch                  Enable
power_limitation_setting                 1100
export_power_limitation                  0
export_power_limitation_value            0
current_transformer_output_current       0
current_transformer_range                0
current_transformer                      0
export_power_limitation_percentage       0
installed_pv_power                       0
power_factor_setting                     0.0
night_svg_switch                         Disable
reactive_power_adjustment_mode           Enable Q(U)
reactive_power_percentage_setting        65535
power_limitation_adjustment              11.0
reactive_power_adjustment                65535
pid_recovery                             Disable
anti_pid                                 0
fullday_pid_suppression                  0
export_to_grid                           0
import_from_grid                         1593
is_running                               True
timestamp                                2021-12-24 13:16:10
```

What I also find a bit strange is that pvoutput is showing 0 values for power generated. Shouldn't this data come be in `v2` which is mapped to the `total_active_power` register and has the right info?

bohdan-s, 2021-12-24:
API:
```
        Parameter   Field               Required    Format      Unit    Example     Donation
        v1          Energy Generation   No          number      wh      10000
        v2          Power Generation    No          number      watts   2000
        v3          Energy Consumption  No          number      wh      10000
        v4          Power Consumption   No          number      watts   2000
```

Config:
```
      - name: v1    # Energy Generation
        register: daily_power_yields
        multiple: 1000
      - name: v2    # Power Generation
        register: total_active_power
      - name: v4    # Power Consumption
        register: load_power
```

Notes from API spec:
_Power can energy values can be derived from one another. When a system sends only power values v2 and v4 the corresponding energy values are automatically calculated.

Similarly when only energy values v1 and v3 are sent, PVOutput will calculate the average power._

Since you are only sending values for v1 and v2 it can't calculate the values correct.

I have just pushed a commit to add manual_load: True to the config.yaml file. If you set that it will manually calculate the load_power (total_active_power + meter_power).

Let me know if that fixes the issue :)

gservat, 2021-12-24:
That looks much better @bohdan-s! Seeing some proper values in pvoutput.org now:

[image]

Thanks again for your help!!

closed 2021-12-25 by gservat