bohdan-s/SunGather #12: Setup

benni336, 2022-01-19:
Hi,
I am just setting up my stuff and I would like to support this work with some documentation work, coding what ever. But currently I am struggling a bit.

At first I am just happy with gathering data via Modbus. All the Database setup (mqtt/influx) can come later (if not needed), when the data is flowing. The webpanel is accessable but shows no data.

I enabled console for debug mode and saw that connection times out.
```

Config               Value                    
host                                     192.168.10.3             
connection                               sungrow                  
model                                    SH8.0RT                  
2022-01-19 11:44:48 ERROR    Connection to (192.168.10.3, 502) failed: timed out
2022-01-19 11:44:58 ERROR    Connection to (192.168.10.3, 502) failed: timed out
2022-01-19 11:45:08 ERROR    Connection to (192.168.10.3, 502) failed: timed out
2022-01-19 11:45:08 WARNING  No data returned for read, 5000:100
				Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.10.3:502)]
2022-01-19 11:45:18 ERROR    Connection to (192.168.10.3, 502) failed: timed out
2022-01-19 11:45:18 WARNING  No data returned for hold, 4999:10
				Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.10.3:502)]
2022-01-19 11:45:18 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
```
I checked other IP addresses to resulting in "no route to host". And the router states that 192.168.10.3 was given to "espressif". So I assume that the host must be there somehow. nmap and ping do not show anything. Modbus is enabled.

I am unsure which ethernet jack I have to use for modbus (Wifi Dongle or the one at the converter). And maybe it is also a problem of my firmeware of the converter version. I could see in Wifi http-page that it is V11. 

I would appreciate any helpful hints!
Best

benni336, 2022-01-19:
This issue can be closed.
I could fix it.

First you need to know that I could only make modbus work on the ethernet jack that is build into the converter.

Second you need to know that there seems to be some firewalling that only allows packages from same IP-range.
So everyone who might do some fancy VLAN stuff to shutdown unwilling traffic from IOT-devices, must know that masquerading traffic helps. So after enabling NAT for incoming traffic from my VLAN 1 -> VLAN Sungrow ... it worked like a charm.

I found a typo in the "try to reconnect" message. I am gonna try to locate it and make a change request :)
Thank you a lot for this amazing piece of code. I am still trying to get deeper into it. But up to now I am just happy to see the current data on my screen.

Best
Ben

bohdan-s, 2022-01-19:
Closing, will add clarification to Readme on requirements :)

closed 2022-01-19 by bohdan-s