bohdan-s/SunGather #14: [Proposal] Adding Units to Registers

benni336, 2022-01-20:
In Reference to #1 I would like to continue that work and jump in.

I have a PDF showing all available registers and I would start to add missing ones to `registers.yaml`
But additionally I also figured out that it is hard to guess what kind of data is behind each value without knowing units.
Units might be usefull to hand them down the road (influx db tag or mqtt?) and it might make more sense to add a "unit" field instead of renaming the register by adding something like `[kwh]` (`registername` -> `registername [kWh]`

What do you all think about adding units?

bohdan-s, 2022-01-20:
Hi Bennie,
If you check the readme you will see I've linked the PDFs of registers. 
If you look at the registers.yaml you will see they are all there. 
If you set your level option to 3 in config.yaml you will see them all returned ( if supported by your inverter). 
I haven't added the unit as I haven't seen a need for it. I have another copy with units in it, but haven't published it as it increases the size and memory usage of the script with little benefit. If you think you could use it I can upload it for you :)

benni336, 2022-01-20:
Well I I think having units could be handy. Eg. as tags in Influx DB for easier labeling of axis or for Display in Home Automatisation or in the WebOutput. I don't know if it those 3-4 characters per register really blow up the Script size.

If you have it already implemented, maybe you can push that version to another branch. So we keep main-branch clean and future changes can (in theory) simply be merged. I would like to have a look at it, maybe you are right and I drop the idea again. As my solar system is quite new, I am sometime a bit confused about the different numbers.

bohdan-s, 2022-01-25:
This is done, check v.0.3.0 :)

closed 2022-01-25 by bohdan-s