bohdan-s/SunGather #15: Register Numbers in Webview

benni336, 2022-01-21:
thanks for your work :)
would it be easy to show the register number alongside the name in the web view? it would make debugging a lot faster for me

_Originally posted by @chainsawbike in https://github.com/bohdan-s/SunGather/issues/1#issuecomment-1018170828_

bohdan-s, 2022-01-21:
I should have a new build up next week. Re-writing a bunch of code. I've kept adding new things now need to consolidate and clean it up.

benni336, 2022-01-21:
This would be a great future (maybe together with log level binding)?

Because I currently have a problem. I added register, but they don't seem to show up in consol and webview. :(
... Yes I have adapted scanning. Maybe this feature could help me finding the problem

bohdan-s, 2022-01-21:
Yup, wait for the new build. Refactoring code to allow it to do more things :) also adding a file export to log to file for those who want it.

benni336, 2022-01-21:
> I should have a new build up next week. Re-writing a bunch of code. I've kept adding new things now need to consolidate and clean it up.

@bohdan-s . I don't want to put any pressure on you. But could you think about just pushing this into an own branch directly? Like a feature branch or a nightly build?

It would make the work for me a bit simpler as I can directly see your changes (they don't need to work or be good). It's a nightly and I could base my development on your last changes as we are in different time zones and seem to have similar ideas and I am currently heavily working on code.

bohdan-s, 2022-01-21:
Sure,
I'm just working on a new build environment to support Raspberry Pi and home assistant then I'll set up a nightly.

benni336, 2022-01-21:
Why is raspberry Pi currently not supported? It's python? Did I miss something?

bohdan-s, 2022-01-25:
v0.3.0 pushed, has addresses listed in console and web-server output now, as well as the unit type. This should help :)

closed 2022-01-25 by bohdan-s