bohdan-s/SunGather #17: export_power / total_active_power out of range (?)

benni336, 2022-01-21:
Hi,
on a cloudy day with very low power generation (like 10-20W in the WiFi App)
export_power and total_active_power seem to show "wrong" values.
```
total_active_power | 65493
export_power | 65076
```

At first I thought it's an out of range thing (like S32 vs U32). According to the guide they should be both S32. But that didn't help.
I didn't yet dive into the code to figure out where it might come from. Maybe it is also an error within the converter.

First wanted to ask if someone here saw something similar.

bohdan-s, 2022-01-21:
What model do you have?
It seems they are returning -ve values because you are drawing from the grid

benni336, 2022-01-22:
Sh10rt and sh8rt.

bohdan-s, 2022-01-25:
@benni336 Could you try adding "indicator: True" to the total_active_power and export_power settings in the registers.yaml file?
This will check the next register for a value of 0xFF and if so convert -ve to +ve value. If thats the case then your inverter is going -ve. I am guessing when drawing from the battery? If so I will need a way to flag that, let me look at the PDF as well and see if there are better registers to be using for the Hybrid inverters.

benni336, 2022-01-25:
No it has nothing to do with the battery. You can see this for inverter with battery (sh8.0rt) and also without battery (sh10rt)

bohdan-s, 2022-03-03:
v0.3.4 should resolve this issue. It was signed values being treated as unsigned.
Please test and let me know
Same as https://github.com/bohdan-s/SunGather/issues/28

bohdan-s, 2022-03-07:
@benni336 could you test and let me know if this is resolved?