bohdan-s/SunGather #18: PVOutput sumbissions stop at midnight

MarkGardiner, 2022-01-25:
Hello,

Thank you! Things are going well with my Home Assistant integration and I am loving this script!

I have seen some odd behaviour the past couple of days after updating my script where the PVOut sumbissions start failing after midnight and I get nothing for the next day. My logs look a little like:

> Jan 25 00:00:16 [] python3[9182]: 2022-01-25 00:00:16 WARNING  No data returned for read, 5000:100
> Jan 25 00:00:16 [] python3[9182]: #011#011#011#011Modbus Error: [Connection] Connection Failed: 301:I18N_COMMON_READ_FAILED
> Jan 25 00:00:16 [] python3[9182]: 2022-01-25 00:00:16 WARNING  No data returned for read, 5100:100
> Jan 25 00:00:16 [] python3[9182]: #011#011#011#011Modbus Error: [Connection] Connection Failed: 301:I18N_COMMON_READ_FAILED
> Jan 25 00:00:16 [] python3[9182]: 2022-01-25 00:00:16 WARNING  No data returned for hold, 4999:10
> Jan 25 00:00:16 [] python3[9182]: #011#011#011#011Modbus Error: [Connection] Connection Failed: 301:I18N_COMMON_READ_FAILED
> Jan 25 00:00:16 [] python3[9182]: 2022-01-25 00:00:16 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
> Jan 25 00:02:49 [] python3[9182]: 2022-01-25 00:02:49 ERROR    PVOutput: Upload Failed; 400 Message; b'Bad request 400: Moon Powered'
> Jan 25 00:03:18 [] python3[9182]: 2022-01-25 00:03:18 ERROR    PVOutput: Upload Failed; 400 Message; b'Bad request 400: Moon Powered'
> .....
> Jan 25 00:17:19 [] python3[9182]: 2022-01-25 00:17:19 ERROR    PVOutput: Upload Failed; 400 Message; b'Bad request 400: Moon Powered'
> ...
> Jan 25 00:32:20 [] python3[9182]: 2022-01-25 00:32:20 ERROR    PVOutput: Upload Failed; 400 Message; b'Bad request 400: Batch size greater than 30 records [60]'
> ...
> Jan 25 00:48:21 [] python3[9182]: 2022-01-25 00:48:21 ERROR    PVOutput: Upload Failed; 403 Message; b'Forbidden 403: Exceeded 60 requests per hour'
> Jan 25 00:48:51 [] python3[9182]: 2022-01-25 00:48:51 ERROR    PVOutput: Upload Failed; 413 Message; b''
> ...
> Jan 25 02:02:55 [] python3[9182]: 2022-01-25 02:02:55 ERROR    PVOutput: Upload Failed; 413 Message; b''
> Jan 25 02:03:25 [] python3[9182]: 2022-01-25 02:03:25 ERROR    PVOutput: Upload Failed; 400 Message; b'\<html\>\<body\>\<h1\>400 Bad request\</h1\>\nYour browser sent an invalid request.\n\</body\>\</html\>\n\n'
> ...
> Jan 25 04:37:02 [] python3[9182]: 2022-01-25 04:37:02 ERROR    PVOutput: Upload Failed; 400 Message; b'\<html\>\<body\>\<h1\>400 Bad request\</h1\>\nYour browser sent an invalid request.\n\</body\>\</html\>\n\n'
> Jan 25 04:37:32 [] python3[9182]: 2022-01-25 04:37:32 ERROR    PVOutput: Failed to Upload
> Jan 25 04:38:02 [] python3[9182]: 2022-01-25 04:38:02 ERROR    PVOutput: Upload Failed; 400 Message; b'\<html\>\<body\>\<h1\>400 Bad request\</h1\>\nYour browser sent an invalid request.\n\</body\>\</html\>\n

As part of my update I also moved to a new config.yaml and tried uncommenting the voltage register for PVOutput but assume thats not an issue here. After commenting the voltage registers again and restarting it all seems to be fine. I then uncommented the voltage, which seemed to work again but hit the same midnight issue last night. Any ideas?

MarkGardiner, 2022-01-25:
Is the script batching up a bunch of values, one of which PVOutput just wont accept and thats tripping it all up?

bohdan-s, 2022-01-25:
Hi @MarkGardiner long story short, once the clock hits midnight PVOutput stops working until restarting the script. 
I am pushing v0.3.0 now which resolves a LOT of outstanding issues. Just check the example-config against your config as a few little things have changed.

MarkGardiner, 2022-01-25:
@bohdan-s gottit, thank you!

closed 2022-01-25 by MarkGardiner