bohdan-s/SunGather #20: Two scrap values for daily datas

benni336, 2022-01-25:
Hi For SH10RT and SH8.0RT I can confirm that daily values are only updated every 5 min. A scraping time below 5 minutes is therefore "useless". How ever there are othere values that are live values that one might want to scrape, like power_load (consumption of your residencial) mppt voltage, battery power or mppt current.

I want so, that's why I scrape every 30 sec.

If we could confirm this behaviour also for the other inverters, we could fix the 5 min interval for the daily values (or make another config setting for it) and keep the current scraping value for the live values.

Scraping is then performed: if $scrape_daily > $scrape_intervall: every $scrape_intervall a "current_dataset" and every $scrape_daily a daily dataset.
 if $scrape_daily <= $scrape_intervall: every $scrape_intervall a full scrape including current and daily data.

Of course I could also reduce the values with an influx database task. I am not using the mqtt or the pvout. So I don't know if there could be any issues with those "plugins".

bohdan-s, 2022-01-25:
Thanks for the thoughts @benni336 
Live values are updated instantly (e.g. active generation and consumption) so reading every 30 secs works.
You are free to set the scan interval to 360 in the settings if you want to reduce the scan time.

There is no point to trying to scan different registers at different times, I am on avg scanning in 1.5s and that is using HTTP scraping of the wifi dongle (my model does not support direct modbus).
If scan times where up in the 10s+ range then I would think about it, but at the moment it would add complexity at no benefit. 
 
At the moment I am avg all scrapes over the PVOutput upload interval, this gives a much smoother graph in PVOutput.
InfluxBD and MQTT have no time limits to data uploads.

closed 2022-01-25 by bohdan-s

benni336, 2022-01-26:
> There is no point to trying to scan different registers at different times, I am on avg scanning in 1.5s and that is using HTTP scraping of the wifi dongle (my model does not support direct modbus). [...]

it was not so much about scanning times but more about reducing the data ammount in your database. But you are right. One could also reduce the load by running daily scheduled task.