bohdan-s/SunGather #21: Cron execution

mistaoh, 2022-01-26:
If I would set up Cron execution and run sungather as one shot would it still handle the PV output rate limit even though I basically stop and rerun the script?

Huge thanks, awesome work!

benni336, 2022-01-26:
I have not checked, but what you say does not really make sense.
The data is stored in an object within the script. When you close the script, the state is lost.

When you use cron, how do you stop scraping? Do you use once only scraping? Why do you want to use cron at all?

bohdan-s, 2022-01-26:
@mistaoh there should be no issue running on a cron. If you are worried about PVOuput just set cron to run every 5 mins.

mistaoh, 2022-01-26:
@benni336 That's the answer I was looking for, was unsure if it saved the timestamp outside the script and was able to handle it anyway. The reason for cron was to be sure it would be running, I had some problems with an earlier version that something broke and the script stopped. 

@bohdan-s I would like it to collect the information at the same rate as it's also used in HA. But I'll close the issue, I'm haven't had an problems with the last version so far. D

closed 2022-01-26 by mistaoh

benni336, 2022-01-30:
For the sake of completness, you can either use an Alert in Grafana or InfluxDB (or what ever you are using along the road) to monitor for incoming data. You could monitor the pid of the started process itself.

You could use a dockerised Version of the script with `restart: always`.
 ... to be continued ...

:)

mistaoh, 2022-01-31:
I will look in to the docker version in the future, for now I just run a script that monitors the process and restarts it if's not running as you mention. 

```
#!/bin/bash
process="sungather.py"
directory="/home/johan/SunGather/SunGather"
makerun="/usr/bin/python3 sungather.py"
if ps ax | grep -v grep | grep -v bash | grep --quiet $process
then
#    printf "Process '%s' is running.\n" "$process"
    exit
else
    printf "(Re)Started %s on '%s'\n" "$process" "$(date)"
    cd $directory
    $makerun
fi
exit
```
And crontab entry to run it every minute
```
* * * * * /home/johan/.supervise/run_sungather.sh >> /home/johan/.supervise/sungather.out 2>&1
```