bohdan-s/SunGather #27: MQTT/HA - Zeroing sensors after inverter switches off

bacco007, 2022-01-30:
Is it possible to zero out the sensors once the inverter switches off overnight?  It's causing the energy dashboard to think there is something to calculate.

[image]

The grid usage that was applicable just before the inverter switched off last night has been projected across overnight until the inverter kicked in again this morning (I also had the same with Solar Production, but with a minimal number)

[image]

benni336, 2022-01-30:
Are you sure there is a value? I would rather think that your tool is misinterpreting missing values. In Grafana e.g you can configure if you want endpoints of graphs to be connected. 
Switch to point style instead of line. 

Which tool do you use for display?

bohdan-s, 2022-01-30:
@bacco007 there could be a few things happening here. 
If the inverter is off then nothing gets uploaded. I.e. it’s like the script isn’t running. If you are using MQTT then that means the system might keep reading the last value if it’s not clearer by the MQTT server. If you are averaging data points then the gap will make things weird. 
I did original send 0 instead of not sending anything but that had different issues for people graphing. 

What export are you using? I can add an option to choose to send 0 every scrape if the scrape fails (e.g. inverter is off). And make it a default as off option. 
would that work?

bacco007, 2022-01-30:
I'm taking the sensors that SunGather creates automatically when you enable HA in the MQTT config 

https://github.com/bohdan-s/SunGather/blob/1a2ca4397c21a08534a4fd25d7da1d4efb112c2a/SunGather/config-example.yaml#L67

bohdan-s, 2022-01-30:
Ok, let me add an option for you to MQTT to send 0 for each value rather than not send anything at all.

bohdan-s, 2022-03-07:
# TO DO, sorry I will resolve this

bacco007, 2022-06-19:
Did you have any luck with this one?  I'm getting some pretty ridiculous figures because HA is taking the last known values before the inverter switches off