bohdan-s/SunGather #28: Very high inverter temperature

mistaoh, 2022-02-03:
Just updated to 3.3 but this was also present in 3.1
I just noticed these temperature spikes on the inverter temperature. I also noticed that the inverter never has a negative temperature even though the outside temperature is around -6°C right now.
[image]

I turned on logging to file so I can provide the debug log file if that would help.

benni336, 2022-02-03:
Hi mistaoh,

thanks for reporting. Interesting observation. 

At first I want to point out that those values don’t have to be wrong. 
For my inverters eg the temperature is about 10 degrees hotter than battery temp. Both are sitting next to each other in the even colder garage. So I measured a range of temps from 5( garage, battery) to 15 (inverter). 

A lot of inverters on the market do have fans, as inverters produce several watts of heat, when working.
How much heat is created may depend on solar pv generation, battery usage and load passing the inverter eg when you use the emergency power or your house (or parts of it) is using emergency power by default.  

But anyhow your graph makes me curious and I will compare to mine when I am back at my raw database. 
It might also be caused by a “bad“/“good” desing for the place of the sensor inside the inverter. 

To get more insight, could you please check the values against the Isolar cloud app using local login? If there are different values, than this is a result of a bug within Sungather. But when you have the same values, than it is caused by the inverter itself and we cannot do anything about that as we only can use the values that are provided by the inverter’s interface. Maybe you can correlate the peaks to known events and btw. what is your inverter model?

Best

bohdan-s, 2022-02-04:
@mistaoh Could you confirm the high value for me?
The inverters store registers in a register which is a signed 16 bit integer (according to the doc).
It seems to be stored as an unsigned 16 bit integer.
Non-technical, it should have a value of -32768 to 32767, but instead it is 0 - 65535.
My guess would be is the value you are seeing the "spike" at would be 65535 - the negative temp reading. so 65534 for -1 degrees.
Could you confirm the actual value? I would never expect a temp over 100oC so I can "split" the value, 0-32768 is positive, 32768-65335 is 65535 - register_value.
The negative temp is something I don't have to worry about in Sydney, Australia ;)
Also as @benni336 said, the reading is internal to the inverter so will be warmer than external temp (or cooler if hot day and your model has a fan)

mistaoh, 2022-02-04:
Here are same time period from isolarcloud and home assistant history. I think negative values roll over to maximum values as you mention. 
[image]
 
[image]

While the Inverter is in stand by it seems to just keep the value but as soon as it starts and is cold it heats up fast to a positive temperature and the issue is fixed.

@benni336 I think you misread the numbers. The inverter is definitely not over 6000°C so for sure its wrong. But thanks for the suggestion to verify against isolarcloud.

benni336, 2022-02-04:
Yep,  I misread the numbers. I thought it was 0-5 degrees. 

Maybe temps are calculated in Kelvin 😂😂. 

Ok, no just joking. Obviously it is a wrong range. Who will do the commit?

bohdan-s, 2022-02-04:
I have code already. I’ll push a “test” channel soon since we seem mostly stable ATM

mistaoh, 2022-02-04:
Sounds good! 👍 It's far from critical issue so no point jeopardizing stability for it.

bohdan-s, 2022-03-01:
Sure. Just post the registers and I’ll patch them tomorrow for you :)

benni336, 2022-03-01:
I provided a list of out of range values below.
But generally speaking: We know that all signed values below 32000 are okay and that all signed values above should be negative. I recalculated those values manually for some occasions and it works great.

We further have a list of signed Modbus values which we have all put into register.yaml.
From my point of view I have never seen any "working" (correctly interpreted) negativ values. Only false interpreted values.
Might it be that the signed/unsigned thing in the register.yaml is not working in gernal?

In that case? Can we fix it or wouldn't it be better to apply the patch to all signed values, even though we (currently) do not observe misinterpretations?

Here my findings:
Register | Value
5033 | total_reactive_power | 64530 Var
13010 | export_power|65390 W
13008 | load_power

and already mentioned:
008 | internal_temperature

bohdan-s, 2022-03-03:
v0.3.4 should resolve this issue. It was signed values being treated as unsigned.
Please test and let me know

bohdan-s, 2022-03-07:
@mistaoh could you please test with v0.3.6 and let me know if its resolved?

mistaoh, 2022-03-07:
@bohdan-s
Thanks for the update, I've updated to 0.3.6 and will check. Unfortunately (not really 😊) the weather have improved greatly over here and it's not as cold any longer. But I will have a look to see if I get any negative temperatures if we get some more cold nights.

mistaoh, 2022-03-14:
@bohdan-s
I can confirm that the issue is solved for me.
Last temperature spike was march seven, before the update. (cant show it since it hides all normal temperatures)
[image]
I do still get some wierd negative temperature spikes but i suspect that is something from the inverter. I can see something like it on iSolarcloud as well. Maybe I will look in to some averaging filter if it annoys me.  
I leave it to you to close it, not sure if everything else is confirmed working.