bohdan-s/SunGather #29: HA Addon

ratava, 2022-02-04:
Dev issue for  the new HA Addon for Sungather. 

Current state is working SunGather under Terminal SSH Docker Session. Working Addon install interface. To Do: Complete Dockerfile to install prerequistes for python.

benni336, 2022-02-05:
Hi, can you point out what do you mean by "HA Addon"?

bohdan-s, 2022-02-05:
HA = Home Assistant 
At the moment it does MQTT and auto discovery. But I would like to fully integrate it so it can be run as an add on instead of standalone script / docker

bohdan-s, 2022-03-07:
Sorry, this is my next priority. 
Have to re-write some code to allow it to run in HA/HACS

highground88, 2022-08-29:
How is this coming along? Waiting to try it 😁 Thanks

andres-canello, 2022-08-29:
I'll be happy to test and buy you beer as well :)

bohdan-s, 2022-12-31:
This has been published as a Hass.io Addon, see  https://github.com/bohdan-s/hassio-repository

closed 2022-12-31 by bohdan-s