bohdan-s/SunGather #30: Question: What is 'Inverter Power State' / run_state?

andres-canello, 2022-02-10:
Mine is OFF most of the time.

Btw, great job, looking forward to Home Assistant support via HACS. :)

benni336, 2022-02-10:
Depending on your inverter run state is not yet fully supported. i am working on making it work with hybrid ones and for them the einstanze currently is just some default value except for some cases.

Can’t say anything about the other inverters though.

andres-canello, 2022-02-11:
Thanks, mine is a SG8K-D.

bohdan-s, 2022-02-11:
If you could post the console/web output when you know the inverter is running I’ll look at the registers and see why it’s saying OFF.

bohdan-s, 2022-02-11:
Hmm, will need to learn what “Derating Run” means. Lol
It seems the inverter is running but at a limited capacity. 
I’ll push a patch to flag that as ON since it is running.

andres-canello, 2022-02-11:
For what I've read it decreases production if it hits 250v in a period of time. Initially I thought it was related to temp but it's not.

Andres Canello
Principal Program Manager
Azure AD

bohdan-s, 2022-03-03:
Hi,
I have fixed this in 0.3.4, please test and let me know if it still shows OFF values when it shouldn't

andres-canello, 2022-03-03:
Yep, that worked great. Thanks mate. Where is HACS support in your priority list? :) beer you way it it helps as an incentive

bohdan-s, 2022-03-03:
HACS is my main focus :) 
There are some limitations with how it works (like having a script constantly running).

rark-ha, 2022-04-12:
> Hmm, will need to learn what “Derating Run” means. Lol It seems the inverter is running but at a limited capacity. I’ll push a patch to flag that as ON since it is running.

"Derating Run" is when the solar is producing more than can be consumed or exported. For example, my grid only lets me export 5KW so if I produce 8KW and can't consume 3KW it derates.

You used to publish the work_state_1 as an attribute on one of the HA sensors. I had a template sensor setup to pull the work state, and use it to trigger automations (e.g. when Derating, I would turn on the pool pump and other battery chargers to try and use the free power). Can we get the attribute back, or create a new additional sensor that reports the actual state instead of the binary sensor?

lucahammer, 2022-04-19:
@rark-ha You can add/edit sensors in the config.yaml yourself. I added it like this:
```
  # Publish Registers to MQTT / Home Assistant
  - name: mqtt
    …
    ha_sensors:
      …
      - name: "Work State"
        register: work_state_1
```
Because all sensors get an unit_of_measurement at the moment, Home Assistant doesn't handle it as a discrete value. I am nut sure if your automation needs that. Pull request #48 fixes it.

Back to this issue: My run_state often is incorrectly OFF as well, but I haven't figured out the pattern yet. I will keep an eye on it and report back, once I know more. (SG20RT)

rark-ha, 2022-04-20:
> @rark-ha You can add/edit sensors in the config.yaml yourself. I added it like this:
>
> ```
>   # Publish Registers to MQTT / Home Assistant
>   - name: mqtt
>     …
>     ha_sensors:
>       …
>       - name: [...]

Thank you! I've created the MQTT sensor and now have it linked into the automations (they monitor the state change so the UoM issue doesn't affect it).