bohdan-s/SunGather #31: SG10RT

benni336, 2022-02-26:
Hi everyone,
thank you very much for this great tool.
As you know I am using it for SH-Inverters and it works great.

But know I need to work with an SG-Inverter (SG10RT). The WiNetS WebInterfaces is accessable via ethernet. But I do not really get, how it works now and what I have to adapt?
I thought I need to use http instead of sungrow? Can anyone give me a push into the right direct?

bohdan-s, 2022-02-26:
Yes, just set to http instead of Sungrow. That’s all you need to do. I wrote a full drop in replacement module to imitate the Sungrow module without any changes :)

benni336, 2022-02-27:
Hi,
I have done so, but it did not work. Sungather gave me an error saying he could not connect.
I than activated DEBUG and could see that I got an "Success" Feedback from the initial connect request returning the device information (SG10RT).
The next message was the error from sungather stating that the connect was not(!) successfull.
I have the feeling that he is not understanding the return correctly or the device data.

Sadly I did not save the terminal output. I will try again and give you an update including the log data.
Just out of curiosity: http method would work with an SH10RT too, right? As soon as there is an WiNetS Web-Interface, http is a possibility to scrape the data. Correct?

Thanks for the great work and quick feedback! I will update you later.

bohdan-s, 2022-02-27:
It should work. If you could post the debug that would help. Also check the SungrowModbusWebclient PIP is at 0.2.7

bohdan-s, 2022-02-27:
So the SungrowModbusWebclient authenticates as a user to the web service to get a token, then does a POST to the Modbus form in the web interface, collects the returned data and adds the required header to appear like a normal modbus response. That means the script code only needs to assume it’s getting modbus data, no matter which of the 3 connection types are used.

benni336, 2022-02-28:
Hi bohdan-s,
thanks for your explenations. That's what I have assumed how it works.

I noticed, that I still have the terminal open and can show you some logs. The logs show you version 0.2.7 in line 11.
This is a test run with a **SH**10, but it was the same for the **SG10** one. Anyway it should work with both, right? :)

```
python3 sungather.py -c test.yaml
2022-02-26 16:38:40 INFO     Starting SunGather 0.3.2
2022-02-26 16:38:40 INFO     Loaded config: test.yaml
2022-02-26 16:38:41 INFO     Loaded registers: /home/pi/SunGather/registers.yaml
2022-02-26 16:38:41 INFO     Logging to console set to: DEBUG
2022-02-26 16:38:41 DEBUG    Inverter Config Loaded: {'host': '192.168.0.248', 'port': 502, 'timeout': 10, 'retries': 3, 'slave': 1, 'scan_interval': 30, 'connection': 'http', 'model': None, 'smart_meter': False, 'use_local_time': False, 'log_console': 'DEBUG', 'log_file': 'OFF', 'level': 2}
2022-02-26 16:38:41 DEBUG    Checking Modbus Connection
2022-02-26 16:38:41 INFO     Modbus client is not connected, attempting to reconnect
2022-02-26 16:38:41 INFO     Connection: SungrowModbusWebClient_0.2.7(192.168.0.248:8082)
2022-02-26 16:38:41 DEBUG    Connection to websocket server established: ws://192.168.0.248:8082/ws/home/overview
2022-02-26 16:38:41 DEBUG    {'result_code': 1, 'result_msg': 'success', 'result_data': {'service': 'connect', 'token': '49f0b56d-ea1f-4d5d-af96-d65563156b72', 'uid': 1, 'tips_disable': 1}}
2022-02-26 16:38:41 INFO     Token Retrieved: 49f0b56d-ea1f-4d5d-af96-d65563156b72
2022-02-26 16:38:41 DEBUG    Requesting Device Information
2022-02-26 16:38:41 DEBUG    {'result_code': 1, 'result_msg': 'success', 'result_data': {'service': 'runtime', 'count': 1, 'list': [{'dev_name': 'SH10RT(COM1-001)', 'dev_model': 'SH10RT', 'dev_type': 35, 'dev_procotol': 2, 'today_energy': '--', 'today_energy_unit': 'kWh', 'total_energy': '--', 'total_energy_unit': 'kWh', 'dev_state': '0', 'dev_state_unit': '', 'curr_power': '0.66', 'curr_power_unit': 'kW', 'reactive_power': '-0.01', 'reactive_power_unit': 'kvar'}], 'connect_count': 1, 'off_count': 0}}
2022-02-26 16:38:41 ERROR    Error: Connection to inverter failed: 192.168.0.248:502
Error: Connection to inverter failed: 192.168.0.248:502
Error: Connection to inverter failed: 192.168.0.248:502
pi@raspberrypi:~/SunGather $ client_loop: send disconnect: Broken pipe
```

puijken, 2022-03-05:
I have exactly the same issue with a SG5.0RS - I'm seeing results in the debug log (correct values) but it still shuts down with exactly the same message as benni336 has.

I'm using the docker container - is this running SungrowModbuswebClient 0.3.0 ?

bohdan-s, 2022-03-05:
Hi, it says "modbus" even for HTTP because is emulating the modbus responses. 
What type of inverter is the SG5.0RS?
I think the issue is I don't have the model numbers for that device. I can see SH5.0RS and SG5.0RT. 
Could you request the modbus map from Sungrow support (they only give you it if you have the right model registered).

puijken, 2022-03-05:
OK I will try to get the information - is there anything I can do myself to pull the data? 

It should probably be (almost) identical to the SG5.0RT but the RS is just single phase.

bohdan-s, 2022-03-05:
The HTTP connection uses a model number to serve the request. See appendix 6 in this PDF https://github.com/bohdan-s/Sungrow-Inverter/blob/539714f42da0374dcc79c2ae36c96c08cff89a81/Modbus%20Information/Communication%20Protocol%20of%20PV%20Grid-Connected%20String%20Inverters_V1.1.37_EN.pdf
I have the PV and Hybrid list but neither have the RS model. 
Easiest way to get the model codes is the modbus file from them. If not we can reverse engineer it.

puijken, 2022-03-05:
Hmmm ok I see, good information. I have sent an email to Sungrow - let's see if they reply 👍

bohdan-s, 2022-03-05:
Just updated SungrowModbusWebClient to 0.3.2 to remove the requirement for a static model list (this was on my TO DO and I forgot about it).
Bumped SunGather to 0.3.5 to update requirements and Docker.
Give 0.3.5 a try for me and let me know how it goes :)

puijken, 2022-03-05:
I was silently thinking the same - since I noticed many registers are the same on Sungrow between models. Container keeps running now and seems to report stuff.

From the webinterface:
```
Sungather v0.3.5
Address
Register	Value
----	device_type_code	None
----	run_state	ON
5003	daily_power_yields	0.0 kWh
5004	total_power_yields	46 kWh
5008	internal_temperature	0.0 °C
5019	phase_a_voltage	0.0 V
5113	daily_running_time	0 min
5006	start_stop	Start
----	export_to_grid	0 W
----	import_from_grid	0 W
----	timestamp	2022-3-6 0:37:53
Total 11 registers

Configuration	Value
host	10.0.200.98
port	8082
timeout	10
retries	3
RetryOnEmpty	False
slave	1
model	None
level	1
use_local_time	False
smart_meter	False
connection	http

```

Great stuff! Will check tomorrow how everything reports to MQQT and PVOutput!

bohdan-s, 2022-03-07:
v0.3.6 psuhed, with heaps more registers for Hybrid inverters and support for the SG5.0RS inveter

benni336, 2022-03-07:
Sorry, I am currently short in time. i can not test before weekend

bohdan-s, 2022-03-07:
@benni336  don’t be sorry, you have been a huge help! Anytime is fine

puijken, 2022-03-07:
Got the new modbus map and documentation from Sungrow today:

[TI_20220111_Communication Protocol of PV Grid-Connected String Inverters_V1.1.39_EN.pdf](https://github.com/bohdan-s/SunGather/files/8197724/TI_20220111_Communication.Protocol.of.PV.Grid-Connected.String.Inverters_V1.1.39_EN.pdf)

[Modbus.pdf](https://github.com/bohdan-s/SunGather/files/8197743/Modbus.pdf)

Hope it can help the further development of this application. 
For me with the SG5.0RS - I will update later today after sun is down to the newest version and report back my findings. In any case - the current setup (with my manually modified registers.yaml) is working and reporting fine to PVOutput and MQTT.

puijken, 2022-03-08:
> v0.3.6 psuhed, with heaps more registers for Hybrid inverters and support for the SG5.0RS inveter

Updated and tested - works perfect! Thanks a lot.

closed 2022-03-09 by bohdan-s