bohdan-s/SunGather #33: Please make sure mqtt.py exists in the exports folder

arfrater, 2022-03-13:
Hi
Still getting this error
2022-03-13 12:51:45 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
                             Please make sure mqtt.py exists in the exports folder
Failed loading export: 'NoneType' object has no attribute 'replace'
                             Please make sure mqtt.py exists in the exports folder
Checked MQTT settings ok Checked the folder exports and mqtt.py exists ?

bohdan-s, 2022-03-13:
Thanks. I’m looking into it now. I think it’s a change in the MQTT library.

arfrater, 2022-03-19:
> Thanks. I’m looking into it now. I think it’s a change in the MQTT library.

Any luck with this issue yet ??

bohdan-s, 2022-03-19:
@arfrater sorry. I'm terrible at updating tickets. 
I've had a good play, and can't get mine to fail this week. 
Could you post a debug log? Only need the first 30 secs or so.

arfrater, 2022-03-19:
Hi @bohdan-s all good. Err stupid question how to get the script to write to a log file ??"

bohdan-s, 2022-03-19:
Make sure you are running the latest version, then set "log_file: DEBUG" in the inverter settings of config.yaml. 

If you are running docker you will also need to mount /logs/ dir (see the readme).

bohdan-s, 2022-03-19:
You could also just set "log_console: DEBUG" and copy paste if you have access to the console / running the script.

arfrater, 2022-03-19:
OK in the morning (night time now solar not operating)

arfrater, 2022-03-19:
Got this tonight 
`2022-03-19 21:48:34 INFO     Loading Export: exports\webserver
2022-03-19 21:48:34 INFO     Webserver: Configured
2022-03-19 21:48:34 INFO     Loading Export: exports\influxdb
2022-03-19 21:48:34 ERROR    InfluxDB: Configured to use total_active_power but not configured to scrape this register
InfluxDB: Configured to use total_active_power but not configured to scrape this register
2022-03-19 21:48:34 INFO     Loading Export: exports\mqtt
2022-03-19 21:48:34 ERROR    Failed loading export: 'NoneType' object has no attribute 'replace'
                             Please make sure mqtt.py exists in the exports folder
Failed loading export: 'NoneType' object has no attribute 'replace'
                             Please make sure mqtt.py exists in the exports folder
2022-03-19 21:48:34 DEBUG    Checking Modbus Connection
2022-03-19 21:48:34 INFO     Modbus, Connecting new session
2022-03-19 21:48:37 ERROR    Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
2022-03-19 21:48:37 DEBUG    Scraping: read, 5000:100
2022-03-19 21:48:37 DEBUG    load_registers: read, 5000:100
2022-03-19 21:48:40 ERROR    Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
2022-03-19 21:48:40 WARNING  No data returned for read, 5000:100
No data returned for read, 5000:100
2022-03-19 21:48:40 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.101:502)]')
2022-03-19 21:48:40 DEBUG    Scraping: read, 5100:100
2022-03-19 21:48:40 DEBUG    load_registers: read, 5100:100
2022-03-19 21:48:43 ERROR    Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
2022-03-19 21:48:43 WARNING  No data returned for read, 5100:100
No data returned for read, 5100:100
2022-03-19 21:48:43 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.101:502)]')
2022-03-19 21:48:43 DEBUG    Scraping: hold, 4999:10
2022-03-19 21:48:43 DEBUG    load_registers: hold, 4999:10
2022-03-19 21:48:46 ERROR    Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
Connection to (192.168.1.101, 502) failed: [Errno 113] No route to host
2022-03-19 21:48:46 WARNING  No data returned for hold, 4999:10
No data returned for hold, 4999:10
2022-03-19 21:48:46 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.101:502)]')
2022-03-19 21:48:46 DEBUG    Scraping: hold, 5009:10
2022-03-19 21:48:46 DEBUG    load_registers: hold, 5009:10
^CTraceback (most recent call last):
  File "/root/SunGather/SunGather/sungather.py", line 566, in <module>
    main()
  File "/root/SunGather/SunGather/sungather.py", line 530, in main
    success = inverter.scrape()
  File "/root/SunGather/SunGather/sungather.py", line 296, in scrape
    if not self.load_registers(range.get('type'), int(range.get('start')), int(range.get('range'))):
  File "/root/SunGather/SunGather/sungather.py", line 176, in load_registers
    rr = self.client.read_holding_registers(start,count=count, unit=self.inverter_config['slave'])
  File "/usr/local/lib/python3.9/dist-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/dist-packages/pymodbus/client/sync.py", line 106, in execute
    if not self.connect():
  File "/usr/local/lib/python3.9/dist-packages/SungrowModbusTcpClient/SungrowModbusTcpClient.py", line 50, in connect
    result = ModbusTcpClient.connect(self)
  File "/usr/local/lib/python3.9/dist-packages/pymodbus/client/sync.py", line 209, in connect
    self.socket = socket.create_connection(
  File "/usr/lib/python3.9/socket.py", line 831, in create_connection
    sock.connect(sa)
KeyboardInterrupt
`

arfrater, 2022-03-19:
OK interesting if I use the  model: "SG5K-D"  in the config.yaml it connects to the MQTT server ??

SamKBEL, 2022-11-06:
I had the same issue. 

The problem is that my system is not autodetected. I have a SG4.0RS and it is not in the list https://github.com/bohdan-s/SunGather#supported.

I corrected by manually set the device name.

May be add a check in the device name and add a error message.

Thank for your work

babstar99, 2023-02-16:
I'm having the same issue.  Previously working.  Also have running on a stand alone docker image that works.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Skipping auto MQTT set up, please ensure MQTT settings are configured in /share/SunGather/config.yaml
2023-02-16 17:04:19 INFO     Starting SunGather 0.5.1
2023-02-16 17:04:19 INFO     Need Help? https://github.com/bohdan-s/SunGather
2023-02-16 17:04:19 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2023-02-16 17:04:19 INFO     Loaded config: /share/SunGather/config.yaml
2023-02-16 17:04:20 INFO     Loaded registers: registers-sungrow.yaml
2023-02-16 17:04:20 INFO     Registers file version: 0.2.3
2023-02-16 17:04:20 INFO     Logging to console set to: DEBUG
2023-02-16 17:04:20 DEBUG    Inverter Config Loaded: {'host': '192.168.0.75', 'port': 502, 'timeout': 10, 'retries': 3, 'slave': 1, 'scan_interval': 30, 'connection': 'http', 'model': None, 'smart_meter': True, 'use_local_time': False, 'log_console': 'DEBUG', 'log_file': 'OFF', 'level': 1}
2023-02-16 17:04:20 INFO     Loading SungrowClient 0.1.0
2023-02-16 17:04:20 DEBUG    Checking Modbus Connection
2023-02-16 17:04:20 INFO     Modbus client is not connected, attempting to reconnect
2023-02-16 17:04:20 INFO     Connection: SungrowModbusWebClient_0.3.2(192.168.0.75:8082)
2023-02-16 17:04:20 DEBUG    Connection to websocket server established: ws://192.168.0.75:8082/ws/home/overview
2023-02-16 17:04:20 DEBUG    {'result_code': 1, 'result_msg': 'success', 'result_data': {'service': 'connect', 'token': 'c902e1e4-c65a-4fb1-b9bf-97a80599826e', 'uid': 1, 'tips_disable': 0}}
2023-02-16 17:04:20 INFO     Token Retrieved: c902e1e4-c65a-4fb1-b9bf-97a80599826e
2023-02-16 17:04:20 DEBUG    Requesting Device Information
2023-02-16 17:04:20 DEBUG    {'result_code': 1, 'result_msg': 'success', 'result_data': {'service': 'devicelist', 'list': [{'id': 1, 'dev_id': 1, 'dev_code': 9267, 'dev_type': 21, 'dev_procotol': 2, 'inv_type': 0, 'dev_sn': 'A2291358417', 'dev_name': 'SG10RT(COM1-001)', 'dev_model': 'SG10RT', 'port_name': 'COM1', 'phys_addr': '1', 'logc_addr': '1', 'link_status': 1, 'init_status': 1, 'dev_special': '0', 'list': []}], 'count': 1}}
2023-02-16 17:04:20 DEBUG    Retrieved: dev_type = 21, dev_code = 9267
2023-02-16 17:04:23 DEBUG    load_registers: read, 4999:1
2023-02-16 17:04:23 DEBUG    Current transaction state - IDLE
2023-02-16 17:04:23 DEBUG    Running transaction 1
2023-02-16 17:04:23 DEBUG    SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x87 0x0 0x1
2023-02-16 17:04:23 DEBUG    New Transaction state 'SENDING'
2023-02-16 17:04:23 DEBUG    param_type: 0, start_address: 5000, count: 1, dev_id: 1
2023-02-16 17:04:23 DEBUG    Calling: http://192.168.0.75/device/getParam?dev_id=1&dev_type=21&dev_code=9267&type=3&param_addr=5000&param_num=1&param_type=0&token=c902e1e4-c65a-4fb1-b9bf-97a80599826e&lang=en_us&time123456=1676527463
2023-02-16 17:04:23 DEBUG    Starting new HTTP connection (1): 192.168.0.75:80
2023-02-16 17:04:33 WARNING  No data returned for read, 4999:1
No data returned for read, 4999:1
2023-02-16 17:04:33 DEBUG    Modbus Error: [Connection] HTTP Request failed: HTTPConnectionPool(host='192.168.0.75', port=80): Max retries exceeded with url: /device/getParam?dev_id=1&dev_type=21&dev_code=9267&type=3&param_addr=5000&param_num=1&param_type=0&token=c902e1e4-c65a-4fb1-b9bf-97a80599826e&lang=en_us&time123456=1676527463 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f066b38d570>, 'Connection to 192.168.0.75 timed out. (connect timeout=10)'))')
2023-02-16 17:04:33 INFO     Model detection failed, please set model in config.py
2023-02-16 17:04:33 DEBUG    load_registers: read, 4989:10
2023-02-16 17:04:33 DEBUG    Current transaction state - SENDING
2023-02-16 17:04:33 DEBUG    Running transaction 2
2023-02-16 17:04:33 DEBUG    SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x7d 0x0 0xa
2023-02-16 17:04:33 DEBUG    New Transaction state 'SENDING'
2023-02-16 17:04:33 DEBUG    param_type: 0, start_address: 4990, count: 10, dev_id: 1
2023-02-16 17:04:33 DEBUG    Calling: http://192.168.0.75/device/getParam?dev_id=1&dev_type=21&dev_code=9267&type=3&param_addr=4990&param_num=10&param_type=0&token=c902e1e4-c65a-4fb1-b9bf-97a80599826e&lang=en_us&time123456=1676527473
2023-02-16 17:04:33 DEBUG    Starting new HTTP connection (1): 192.168.0.75:80
2023-02-16 17:04:43 WARNING  No data returned for read, 4989:10
No data returned for read, 4989:10
2023-02-16 17:04:43 DEBUG    Modbus Error: [Connection] HTTP Request failed: HTTPConnectionPool(host='192.168.0.75', port=80): Max retries exceeded with url: /device/getParam?dev_id=1&dev_type=21&dev_code=9267&type=3&param_addr=4990&param_num=10&param_type=0&token=c902e1e4-c65a-4fb1-b9bf-97a80599826e&lang=en_us&time123456=1676527473 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f066b38e0e0>, 'Connection to 192.168.0.75 timed out. (connect timeout=10)'))')
2023-02-16 17:04:43 INFO     Serial detection failed, please set serial number in config.py
[... cut ...]
2023-02-16 17:04:43 DEBUG    SEND: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x26
2023-02-16 17:04:43 DEBUG    New Transaction state 'SENDING'
2023-02-16 17:04:43 DEBUG    param_type: 0, start_address: 5001, count: 38, dev_id: 1
2023-02-16 17:04:43 DEBUG    Calling: http://192.168.0.75/device/getParam?dev_id=1&dev_type=21&dev_code=9267&type=3&param_addr=5001&param_num=38&param_type=0&token=c902e1e4-c65a-4fb1-b9bf-97a80599826e&lang=en_us&time123456=1676527483
2023-02-16 17:04:43 DEBUG    Starting new HTTP connection (1): 192.168.0.75:80