bohdan-s/SunGather #34: SH6.0RT current/daily load (export_to/import_from grid)

danielHin, 2022-03-16:
Hi, 

@bohdan-s  thanks again for the fast reply on discord. As discussed, SunGather doesn't print out the current consumption/load. You mentioned in the  chat that the hybrid model seem to use different registers which causes the missing data. 

It seems 13008, 13010 and 5216, 5218 to be the data I'm searching for

Following the level 3 log with Smart Meter disabled. 

I'm having a SH6.0RT connected to a BYD Battery and a external Smart Meter. 

Please let me know if you need me to test or I can help in any way. 

```
$ python3 sungather.py
2022-03-17 00:20:26 INFO     Starting SunGather 0.3.7
2022-03-17 00:20:26 INFO     Loaded config: config.yaml
2022-03-17 00:20:26 INFO     Loaded registers: /src/github/SunGather/SunGather/registers.yaml
2022-03-17 00:20:26 INFO     Registers file version: 0.2.0
+----------------------------------------------+
| Inverter Configuration Settings              |
+----------------------------------------------+
| Config             | Value                   |
+--------------------+-------------------------+
| host               | 10.77.8.243             |
| port               | 8082                    |
| timeout            | 10                      |
| retries            | 3                       |
| RetryOnEmpty       | False                   |
| slave              | 1                       |
| model              | SH6.0RT                 |
| level              | 3                       |
| use_local_time     | False                   |
| smart_meter        | False                   |
| connection         | http                    |
+----------------------------------------------+
2022-03-17 00:20:35 WARNING  No data returned for read, 7012:25
No data returned for read, 7012:25
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| 5000    | device_type_code                    | SH6.0RT              |
| ----    | run_state                           | ON                   |
| 4950    | protocol_number                     | 12288                |
| 4952    | protocol_version                    | 4352                 |
| 4954    | arm_software_version                | 16722                |
| 4969    | dsp_software_version                | 19780                |
| 4990    | serial_number                       | xxxxx                |
| 5001    | nominal_active_power                | 6.0 kW               |
| 5002    | output_type                         | 3P4L                 |
| 5003    | daily_power_yields                  | 0.0 kWh              |
| 5004    | total_power_yields                  | 0.0 kWh              |
| 5006    | total_running_time                  | 0 h                  |
| 5008    | internal_temperature                | 27.5 °C              |
| 5009    | total_apparent_power                | 0 VA                 |
| 5011    | mppt_1_voltage                      | 0.0 V                |
| 5012    | mppt_1_current                      | 0.0 A                |
| 5013    | mppt_2_voltage                      | 0.0 V                |
| 5014    | mppt_2_current                      | 0.0 A                |
| 5015    | mppt_3_voltage                      | 0.0 V                |
| 5016    | mppt_3_current                      | 0.0 A                |
| 5017    | total_dc_power                      | 0 W                  |
| 5019    | phase_a_voltage                     | 226.2 V              |
| 5020    | phase_b_voltage                     | 226.2 V              |
| 5021    | phase_c_voltage                     | 226.8 V              |
| 5022    | phase_a_current                     | 0.0 A                |
| 5023    | phase_b_current                     | 0.0 A                |
| 5024    | phase_c_current                     | 0.0 A                |
| 5031    | total_active_power                  | 0 W                  |
| 5033    | total_reactive_power                | 0 Var                |
| 5035    | power_factor                        | 0.0                  |
[... 116 lines cut ...]
| 13118   | min_cell_temp                       | 0 °C                 |
| 5006    | start_stop                          | Start                |
| 5007    | power_limitation_switch             | Enable               |
| 5008    | power_limitation_setting            | 100.0 %              |
| 5010    | export_power_limitation             | 0                    |
| 5011    | export_power_limitation_value       | 0                    |
| 5012    | current_transformer_output_current  | 0 A                  |
| 5013    | current_transformer_range           | 0 A                  |
| 5014    | current_transformer                 | 0                    |
| 5015    | export_power_limitation_percentage  | 0.0 %                |
| 5016    | installed_pv_power                  | 0.0 KW               |
| 5019    | power_factor_setting                | 0.0                  |
| 5035    | night_svg_switch                    | 0                    |
| 5036    | reactive_power_adjustment_mode      | Enable Q(U)          |
| 5037    | reactive_power_percentage_setting   | 0.0 %                |
| 5039    | power_limitation_adjustment         | 0.0 kW               |
| 5040    | reactive_power_adjustment           | 0.0 kVar             |
| 5041    | pid_recovery                        | 0                    |
| 5042    | anti_pid                            | 0                    |
| 5043    | fullday_pid_suppression             | 0                    |
| ----    | export_to_grid                      | 0 W                  |
| ----    | import_from_grid                    | 0 W                  |
| ----    | timestamp                           | 2022-3-17 1:20:33    |
+----------------------------------------------------------------------+
```

benni336, 2022-03-17:
Hi @danielHin ,
I have set up a full functional set of registers in my fork of sungather ( https://github.com/benni336/SunGather/tree/main/SunGather ). 
Unfortunately I can currently not push to bodhan's repository as bodhan is going to release some more code in near future. So he asked me to hold my changes back. I will bring my changes to this repo here as soon as bodhan will accept them then.

You have two options. Either checkout my full code or just the my register.yaml with the current version of sungather you have already installed. I would prefer latter option, as I am not sure if I have already applied the negative value patch.

I can confirm that all relevant registers are working. You do not need to scrape all known registers.

bohdan-s, 2022-03-17:
Those registers all should be in the lastest version of SunGather. Let me know if you spot any missing

danielHin, 2022-03-17:
> Those registers all should be in the lastest version of SunGather. Let me know if you spot any missing
So it already should work or is your comment related to @benni336 comment?

benni336, 2022-03-17:
@danielHin  : I think the comment was meant for me.
Please checkout SH_Inverter register.yaml ( [here](https://github.com/benni336/SunGather/blob/SH_Inverters/SunGather/registers.yaml) ) of my SH_inverter branch. I just checked out of curiousity and when I interpreted to result of my pull request correctly, there is a bunch missing in bohdan-s version.

Unfortunately I never know when it is the right time to file a pull request (I have some more than just the register.yaml). When I do, bodhan-s has already something new in the queue and HEAD has changes faster than I can rebase. When I am lazy and wait, bodhan seem to be lazy too :D

bohdan-s, 2022-03-17:
@benni336 about 95% of your fork has been merged in :) most diff is extra spaces etc...
If you could do some pull requests but limit it per file then we should be able to merge them all :)
I'll hold off any big changes until you do your pull requests.

benni336, 2022-03-22:
@bohdan-s : I have done the rebase work for the register.yaml. and I will move forward. Sorry for beeing absent for a longer time.

I need to apologize also because I claimed bohdhan-s register.yaml is not up to date. In fact it is.
So thanks for merging my stuff into your one (obviously by hand?).
I have added the missing part with pull request #37 .
Best
ben

benni336, 2022-03-28:
@danielHin : even though my change request is not yet accepted, could we resolve your issue, or do you still have any problems?
Best

danielHin, 2022-03-29:
Hey, 

sorry should have tried it earlier..
 
I just gave it a try with your register.yml. import_from_grid is calculated now, but its actually my current export to grid. Seems to be swapped? I'm comparing to isolarcloud output. Also register 5216 (export power) is the same value as import_from_grid.

```
Logged 159 registers to Console
2022-03-29 13:20:46 WARNING  No data returned for read, 7012:25
No data returned for read, 7012:25
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| 5000    | device_type_code                    | SH6.0RT              |
| ----    | run_state                           | ON                   |
| 4950    | protocol_number                     | 12288                |
| 4952    | protocol_version                    | 4352                 |
| 4954    | arm_software_version                | 16722                |
| 4969    | dsp_software_version                | 19780                |
| 4990    | serial_number                       | xxxxx                |
| 5001    | nominal_active_power                | 6.0 kW               |
| 5002    | output_type                         | 3P4L                 |
| 5003    | daily_power_yields                  | 8.8 kWh              |
| 5004    | total_power_yields                  | 0.0 KWh              |
| 5006    | total_running_time                  | 0 h                  |
| 5008    | internal_temperature                | 41.9 °C              |
| 5009    | total_apparent_power                | 3149 VA              |
| 5011    | mppt_1_voltage                      | 409.6 V              |
| 5012    | mppt_1_current                      | 3.7 A                |
| 5013    | mppt_2_voltage                      | 402.0 V              |
| 5014    | mppt_2_current                      | 4.2 A                |
| 5015    | mppt_3_voltage                      | 0.0 V                |
| 5016    | mppt_3_current                      | 0.0 A                |
| 5017    | total_dc_power                      | 3223 W               |
| 5019    | phase_a_voltage                     | 227.7 V              |
| 5020    | phase_b_voltage                     | 225.6 V              |
| 5021    | phase_c_voltage                     | 225.2 V              |
| 5022    | phase_a_current                     | 4.7 A                |
| 5023    | phase_b_current                     | 4.7 A                |
| 5024    | phase_c_current                     | 4.7 A                |
| 5031    | total_active_power                  | 3165 W               |
| 5033    | total_reactive_power                | -3 Var               |
| 5035    | power_factor                        | 1.0                  |
| 5036    | grid_frequency                      | 49.99 Hz             |
| 5038    | work_state_1                        | 0                    |
| 5049    | nominal_reactive_power              | 3.0 kVar             |
| 5101    | daily_direct_energy_consumption     | 3.6 kWh              |
| 5103    | total_direct_energy_consumption     | xxxxxx kWh           |
| 5113    | daily_running_time                  | 0 min                |
| 5115    | mppt_4_voltage                      | 0.0 V                |
| 5116    | mppt_4_current                      | 0.0 A                |
| 5117    | mppt_5_voltage                      | 0.0 V                |
| 5118    | mppt_5_current                      | 0.0 A                |
| 5119    | mppt_6_voltage                      | 0.0 V                |
| 5120    | mppt_6_current                      | 0.0 A                |
| 5121    | mppt_7_voltage                      | 0.0 V                |
| 5122    | mppt_7_current                      | 0.0 A                |
| 5123    | mppt_8_voltage                      | 0.0 V                |
| 5124    | mppt_8_current                      | 0.0 A                |
| 5128    | monthly_power_yields                | 0.0 kWh              |
| 5130    | mppt_9_voltage                      | 0.0 V                |
| 5131    | mppt_9_current                      | 0.0 A                |
| 5132    | mppt_10_voltage                     | 0.0 V                |
| 5133    | mppt_10_current                     | 0.0 A                |
| 5134    | mppt_11_voltage                     | 0.0 V                |
| 5135    | mppt_11_current                     | 0.0 A                |
| 5136    | mppt_12_voltage                     | 0.0 V                |
| 5137    | mppt_12_current                     | 0.0 A                |
| 5146    | negative_voltage_to_the_ground      | 0.0 V                |
| 5147    | bus_voltage                         | 615.0 V              |
[... 81 lines cut ...]
| 5007    | power_limitation_switch             | Enable               |
| 5008    | power_limitation_setting            | 100.0 %              |
| 5010    | export_power_limitation             | 0                    |
| 5011    | export_power_limitation_value       | 0                    |
| 5012    | current_transformer_output_current  | 0 A                  |
| 5013    | current_transformer_range           | 0 A                  |
| 5014    | current_transformer                 | 0                    |
| 5015    | export_power_limitation_percentage  | 0.0 %                |
| 5016    | installed_pv_power                  | 0.0 KW               |
[... cut ...]
| 5041    | pid_recovery                        | 0                    |
| 5042    | anti_pid                            | 0                    |
| 5043    | fullday_pid_suppression             | 0                    |
| ----    | export_to_grid                      | 0 W                  |
| ----    | import_from_grid                    | 2897 W               |
| ----    | timestamp                           | 2022-3-29 13:20:46   |
+----------------------------------------------------------------------+
Logged 159 registers to Console

```

danielHin, 2022-07-08:
fixed #60 

@bohdan-s @benni336 
I hope I didn't miss something by assuming that total_active_power and load_power_hybrid is the right values for the calculations. 

In case the Battery is empty and no DC power is generated, total_active_power should output 0. Thtas my assumption

danielHin, 2022-07-26:
fixed by #60

closed 2022-07-26 by danielHin