bohdan-s/SunGather #40: index out of range

yoursnack, 2022-03-30:
Hi I'm having an issue to read values from an SG40CX. I don't know if I should paste here but just be sure, this is my output on pastebin : https://pastebin.com/SqD1Ep3p

It seems to be reading some values but the output is always 0 or nothing at all. Included here the output of the webserver: https://pastebin.com/n8LkGWP2

Greetings and I hope we can work this out together!

bohdan-s, 2022-03-30:
Looks like you have the “encrypted” modbus. Change the setting:
connection: sungrow

yoursnack, 2022-03-31:
Hi, thanks for your quick reply. However it gives me practically the same output.

https://pastebin.com/mKh04WpM

Even if I try port 8082, it can connect, send but gets no reply

bohdan-s, 2022-03-31:
Remove the port option. 
connection: http
;) third time is a charm right?

yoursnack, 2022-03-31:
Nope, that also did not work.
https://pastebin.com/Cc8U90yJ
I did a port scan on it and I guess port 80 is not open at all. Is there anything I can do to help find out what it is so others will not be stuck in the future?

Port Scan TCP:
Starting Nmap 7.80 ( https://nmap.org ) at 2022-03-31 12:24 WEST
Nmap scan report for 172.16.2.198
Host is up (0.14s latency).
Not shown: 998 closed ports
PORT     STATE SERVICE
443/tcp  open  https
8082/tcp open  blackice-alerts

Port scan UDP:
Starting Nmap 7.80 ( https://nmap.org ) at 2022-03-31 12:25 WEST
Nmap scan report for 172.16.2.198
Host is up (0.16s latency).
Not shown: 999 closed ports
PORT     STATE SERVICE
5353/udp open  zeroconf
MAC Address: AC:0B:FB:0A:59:88 (Unknown)

bohdan-s, 2022-03-31:
What dongle/Ethernet adapter are you using? Does the data publish to iSolarCloud at the moment?

bjeanes, 2022-07-07:
I am in a similar position. SH5.0RS using a WiNet-S (version `M_WiNet-S_V01_V01_A`) connected both on WLAN and LAN, trying to connect over LAN connection. `connection: http` works, but neither `sungrow` nor `modbus` work.

It eventually prints out 13 registers, but with predominantly zeroed values:

```
2022-07-07 13:44:24 WARNING  Modbus connection failed
Modbus connection failed
2022-07-07 13:44:24 ERROR    index out of range
index out of range
2022-07-07 13:44:24 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2022-07-07 13:44:24 WARNING  Modbus connection failed
Modbus connection failed
2022-07-07 13:44:24 ERROR    index out of range
index out of range
2022-07-07 13:44:24 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2022-07-07 13:44:24 WARNING  Modbus connection failed
Modbus connection failed
2022-07-07 13:44:24 ERROR    index out of range
index out of range
2022-07-07 13:44:24 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
[... 19 lines cut ...]
    self._process(callback)
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2022-07-07 13:44:24 WARNING  Modbus connection failed
Modbus connection failed
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| ----    | device_type_code                    | SH5.0RS              |
| ----    | run_state                           | ON                   |
| 5006    | start_stop                          | Start                |
[... cut ...]
| 5037    | reactive_power_percentage_setting   | 0.0 %                |
| 5039    | power_limitation_adjustment         | 0.0 kW               |
| 5040    | reactive_power_adjustment           | 0.0 kVar             |
| ----    | export_to_grid                      | 0 W                  |
| ----    | import_from_grid                    | 0 W                  |
| ----    | timestamp                           | 2022-7-7 13:44:24    |
+----------------------------------------------------------------------+
Logged 13 registers to Console
```

bohdan-s, 2022-07-07:
@bjeanes the WiNet-S only supports HTTP method. Modbus gives errors or bad data.

bohdan-s, 2022-07-07:
@yoursnack sorry for the delay, you can only have 1 session open via HTTP do the dongle (well you can have more but its unstable).
Try restarting the dongle and connecting via HTTP port 80 and it should work.

bjeanes, 2022-07-07:
> @bjeanes the WiNet-S only supports HTTP method. Modbus gives errors or bad data.

I'm gathering this. Is this likely to be permanent? If so, are you aware of an alternate dongle I can replace to use Modbus directly? I bought this inverter with the intent to send Modbus commands, not just reading registers 😬

bohdan-s, 2022-07-07:
Thats up to Sungrow to fix :( Some German users are saying their firmware (v10) is working on the discord.

We "can" write registers via the HTTP method, I just haven't implemented it.

What registers are you interesting in writing to?

bjeanes, 2022-07-07:
> Thats up to Sungrow to fix :(

I'll email them to add another voice to hassle them about this. 

> Some German users are saying their firmware (v10) is working on the discord.

Interesting... When I visit the IP in browser I am asked for an installer login. Perhaps with those credentials Modbus access can be enabled and their installers did so? (wishful thinking on my part, probably).

> We "can" write registers via the HTTP method, I just haven't implemented it.

Oh! Interesting. That might work then.

> What registers are you interesting in writing to?

I'll have to review my notes from a few months ago when researching this. It's been evicted from my brain while waiting for the install to take place. But the use-case is describable: I want to be able to force charge and discharge programmatically. I'm on a wholesale energy retailer and have a pretty sizeable PV array and battery. While arbitrage is not really long-term profitable, during wholesale price spikes as have been happening a lot in my region (and probably everywhere), selling a % of my stored energy is something I had hoped to make work.

alangibson-philips, 2022-08-14:
Just adding a datapoint. I have a Sungrow SH8.0RT with WiNet-S dongle (firmware version M_WiNet-S_V01_V01_A). Neither modbus nor sungrow connection settings work. However http works perfectly. It even worked when I was logged in to the web app. I was just summarily logged out when connecting with SunGather.

Errors are just a whole lot of 

```
2022-08-14 22:13:43 DEBUG    SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
2022-08-14 22:13:43 DEBUG    New Transaction state 'SENDING'
2022-08-14 22:13:43 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-08-14 22:13:43 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-08-14 22:13:43 DEBUG    RECV: 0x0 0x1 0x0 0x0 0x0 0x2 0x1 0x84 0x2
2022-08-14 22:13:43 DEBUG    Processing: 0x0 0x1 0x0 0x0 0x0 0x2 0x1 0x84 0x2
2022-08-14 22:13:43 DEBUG    Factory Response[132]
2022-08-14 22:13:43 ERROR    index out of range
2022-08-14 22:13:43 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/alangibson/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/alangibson/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/home/alangibson/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
```

bjeanes, 2022-08-14:
@alangibson-philips update the firmware on the WiNet-S and you will get (admittedly somewhat buggy) Modbus connection working.

The `sungrow` type is misleading (perhaps could be renamed) and is for the encrypted modbus which older SG inverters (or dongles?) used. The new SH##.0RS/T don't use it.