bohdan-s/SunGather #43: Inverter Power State incorrectly OFF

rark-ha, 2022-04-12:
After checking the terminal entries and then logging into the web page, I was logged out of the web page (presumably SunGather was doing it's next scrape).

The reading for run_state started showing OFF when it was in fact ON (confirmed via the web page). I restarted the container to get it to show ON again. Capture of the bug below.

[image]

puijken, 2022-05-04:
I can confirm this issue .. I'm using this to parameter in my logic in NodeRED but this morning my values were not reporting -> seemed run_state was still OFF but the inverter was running and producing energy.

danielHin, 2022-07-29:
Hey @rark-ha can you still reproduce this with the latest version of SunGather? there was some work done around this registers, maybe this is fixed now

rark-ha, 2022-07-30:
No I can't reproduce this issue now

closed 2022-07-30 by rark-ha

puijken, 2022-07-31:
Unfortunately the issue still seems to exist ... please find below an output from my NodeRED where you can clearly see the inverter is running but it still reports "run_state" = OFF

Inverter = SG5.0RS

[image]