bohdan-s/SunGather #47: Wrong voltage is sent to pvoutput.org - total_apparent_power instead of  phase_a_voltage

fork04, 2022-04-18:
The good value/variable is sent after script restart then is getting wrong:
[image]

my config is:

      - name: v5                            # Ambient temperature see: https://forum.pvoutput.org/t/what-temperature-is-used-to-calcul>
        register: internal_temperature      #
      - name: v6                            # Voltage
        register: phase_a_voltage           # Read voltage from first Phase

bohdan-s, 2022-07-07:
@fork04 sorry, can you check this is still happening? it was an issue with the registers file which should now be resolved.

lexleite, 2022-10-06:
i think it still happens. I guess this is that same issue with the average. It seems that it is not dividing by the number of events.

lexleite, 2022-10-08:
The problem in the code is on the pvoutput.py on the line regarding the "trim outlier"

cr750, 2022-10-15:
[image]
I downloaded the latest from GitHub last weekend, so Yes it is still happening.
(Though I did keep my old config.yaml)