bohdan-s/SunGather #53: Mapping for running_state

michbeck100, 2022-06-02:
I started using SunGather and I'm thinking to move from modbus4mqtt. Unfortunately it seems that SunGather doesn't support a mapping for `running_state`. 
According to the sungrow documentation this is the mapping:
[image]
So `running_state` actually contains 8 values.

How can I achieve such a mapping?

danielHin, 2022-07-07:
Hi, the merge introduced a bug. 

`Traceback (most recent call last):
  File "sungather.py", line 584, in <module>
    main()
  File "sungather.py", line 552, in main
    export.publish(inverter)
  File "/src/github/SunGather/SunGather/exports/console.py", line 26, in publish
    print("| {:<7} | {:<35} | {:<20} |".format(str(inverter.getRegisterAddress(register)), str(register), str(value) + " " + str(inverter.getRegisterUnit(register))))
  File "sungather.py", line 265, in getRegisterAddress
    return register['address']
KeyError: 'address'
`

michbeck100, 2022-07-07:
No it happens when reading run_state, which is a custom register without address. It used to have a "virtual" address, but commit bc8493fee93e3c3488930ccc0660a5deef1a455d changed this. This is not because of my changes.

michbeck100, 2022-07-07:
this fixes it again: https://github.com/bohdan-s/SunGather/pull/57

danielHin, 2022-07-08:
thy for fixing. 

@michbeck100 @bohdan-s 
another issue I've just saw.  The newly introduced 'import_from_grid' register in the registers file clashes with the custom 'import_from_grid' register and overwrites the value. 

Maybe we rename the running state registers by prefix them with state_ or running_state_ ? 

Alternatively we could rename import_from_grid and export_to_grid to custom_*.

 But its more obvious at lest to me to rename the new running state registers to make it more clear

danielHin, 2022-07-08:
proposed renaming in #59

michbeck100, 2022-07-15:
I think this can be closed now

closed 2022-07-15 by michbeck100