bohdan-s/SunGather #62: Support writing holding registers

michbeck100, 2022-07-15:
I would like to be able to write registers, too. 

My idea would be to listen to special mqtt topics and/or HTTP PUT on the webserver.

jpwise9, 2024-04-04:
Similarly chiming in on on this - i see there is mention of an alpha release with write support on the SunGrowModBusWebClient issue list - but being able to update specific values on the fly would be great.
I'm specifically looking at updating the battery_max_soc (13058) in an attempt to defer full charge until later in the day and maximise export during peak periods.  Separate issue post regarding reading holding registers to follow.

Thx.

Egyras, 2024-12-20:
@bohdan-s any plans for this ? Would be nice to "simulate" power meeter :)