bohdan-s/SunGather #76: System not recognizing SG3K-D

lexleite, 2022-09-12:
System not recognizing SG3K-D

`
admin@RT-AC68U-A2D8:/tmp/mnt/KINGSTON/entware/home/SunGather-main/SunGather# ./sungather.py -v 10 --runonce > log.txt
2022-09-12 13:19:29 INFO     Starting SunGather 0.3.8
2022-09-12 13:19:30 INFO     Loaded config: config.yaml
2022-09-12 13:19:37 INFO     Loaded registers: /tmp/mnt/KINGSTON/entware/home/SunGather-main/SunGather/registers-sungrow.yaml
2022-09-12 13:19:37 INFO     Registers file version: 0.2.0
2022-09-12 13:19:37 INFO     Logging to console set to: DEBUG
2022-09-12 13:19:37 DEBUG    Inverter Config Loaded: {'host': '192.168.1.208', 'port': 502, 'timeout': 10, 'retries': 3, 'slave': 1, 'scan_interval': 30, 'connection': 'sungrow', 'model': None, 'smart_meter': False, 'use_local_time': False, 'log_console': 'WARNING', 'log_file': 'OFF', 'level': 3}
2022-09-12 13:19:37 DEBUG    Checking Modbus Connection
2022-09-12 13:19:37 INFO     Modbus client is not connected, attempting to reconnect
2022-09-12 13:19:37 INFO     Connection: ModbusTcpClient(192.168.1.208:502)
2022-09-12 13:19:37 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 44772)
2022-09-12 13:19:37 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 57834)
2022-09-12 13:19:40 DEBUG    load_registers: read, 4999:1
2022-09-12 13:19:40 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 47555)
2022-09-12 13:19:40 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 50533)
2022-09-12 13:19:40 DEBUG    Current transaction state - IDLE
2022-09-12 13:19:40 DEBUG    Running transaction 1
2022-09-12 13:19:40 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 35795)
2022-09-12 13:19:40 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 53786)
2022-09-12 13:19:40 DEBUG    SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x87 0x0 0x1
2022-09-12 13:19:40 DEBUG    New Transaction state 'SENDING'
2022-09-12 13:19:40 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-09-12 13:19:40 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-09-12 13:19:40 DEBUG    RECV: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x1 0x22
2022-09-12 13:19:40 DEBUG    Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x1 0x22
2022-09-12 13:19:40 DEBUG    Factory Response[ReadInputRegistersResponse: 4]
2022-09-12 13:19:40 DEBUG    Adding transaction 1
2022-09-12 13:19:40 DEBUG    Getting transaction 1
2022-09-12 13:19:40 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-09-12 13:19:40 WARNING  Unknown Type Code Detected: 290
Unknown Type Code Detected: 290
2022-09-12 13:19:40 INFO     Closing Session: ModbusTcpClient(192.168.1.208:502)
2022-09-12 13:19:40 INFO     Loading Export: exports\webserver
2022-09-12 13:19:40 INFO     Webserver: Configured
2022-09-12 13:19:40 INFO     Loading Export: exports\pvoutput
2022-09-12 13:19:40 DEBUG    PVOutput: Get System ; https://pvoutput.org/service/r2/getsystem.jsp, {'X-Pvoutput-Apikey': 'xxxxxxxxxx', 'X-Pvoutput-SystemId': '93119', 'Content-Type': 'application/x-www-form-urlencoded', 'cache-control': 'no-cache'}, 'teams': '1'
2022-09-12 13:19:40 DEBUG    Starting new HTTPS connection (1): pvoutput.org:443
2022-09-12 13:19:41 DEBUG    https://pvoutput.org:443 "POST /service/r2/getsystem.jsp?teams=1 HTTP/1.1" 200 None
2022-09-12 13:19:41 DEBUG    PVOutput: Response; 200 Message; b'CasaFunfzehn,3630,,11,330,JA Solar,1,3000,Sungrow,N,13.0,No,20201209,-12.949314,-38.3846,5;;1618;0'
2022-09-12 13:19:41 INFO     PVOutput: Configured export to CasaFunfzehn every 5 minutes
2022-09-12 13:19:41 DEBUG    Checking Modbus Connection
2022-09-12 13:19:41 INFO     Modbus, Connecting new session
2022-09-12 13:19:41 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 36283)
2022-09-12 13:19:41 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 46697)
2022-09-12 13:19:41 DEBUG    Scraping: read, 4949:50
2022-09-12 13:19:41 DEBUG    load_registers: read, 4949:50
2022-09-12 13:19:41 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 57582)
2022-09-12 13:19:41 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 59253)
2022-09-12 13:19:41 DEBUG    Current transaction state - TRANSACTION_COMPLETE
2022-09-12 13:19:41 DEBUG    Running transaction 2
2022-09-12 13:19:41 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 38093)
2022-09-12 13:19:41 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 41436)
2022-09-12 13:19:41 DEBUG    SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x55 0x0 0x32
2022-09-12 13:19:41 DEBUG    New Transaction state 'SENDING'
2022-09-12 13:19:41 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-09-12 13:19:41 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-09-12 13:19:41 DEBUG    RECV: 0x0 0x2 0x0 0x0 0x0 0x67 0x1 0x4 0x64 0x30 0x0 0x41 0x52 0xf 0x0 0x1 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x41 0x31 0x39 0x30 0x38 0x30 0x35 0x32 0x37 0x35 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-09-12 13:19:41 DEBUG    Processing: 0x0 0x2 0x0 0x0 0x0 0x67 0x1 0x4 0x64 0x30 0x0 0x41 0x52 0xf 0x0 0x1 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x41 0x31 0x39 0x30 0x38 0x30 0x35 0x32 0x37 0x35 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-09-12 13:19:41 DEBUG    Factory Response[ReadInputRegistersResponse: 4]
2022-09-12 13:19:41 DEBUG    Adding transaction 2
2022-09-12 13:19:41 DEBUG    Getting transaction 2
2022-09-12 13:19:41 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-09-12 13:19:42 DEBUG    Scraping: read, 5000:100
2022-09-12 13:19:42 DEBUG    load_registers: read, 5000:100
2022-09-12 13:19:42 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 37117)
2022-09-12 13:19:42 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 55642)
2022-09-12 13:19:42 DEBUG    Current transaction state - TRANSACTION_COMPLETE
[... 312 lines cut ...]
2022-09-12 13:19:48 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 42448)
2022-09-12 13:19:48 DEBUG    Connection to Modbus server established. Socket ('192.168.1.1', 57009)
2022-09-12 13:19:48 DEBUG    SEND: 0x0 0x13 0x0 0x0 0x0 0x6 0x1 0x3 0x32 0xc8 0x0 0x64
2022-09-12 13:19:48 DEBUG    New Transaction state 'SENDING'
2022-09-12 13:19:48 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-09-12 13:19:49 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-09-12 13:19:49 DEBUG    RECV: 0x0 0x13 0x0 0x0 0x0 0x3 0x1 0x83 0x2
2022-09-12 13:19:49 DEBUG    Processing: 0x0 0x13 0x0 0x0 0x0 0x3 0x1 0x83 0x2
2022-09-12 13:19:49 DEBUG    Factory Response[131]
2022-09-12 13:19:49 DEBUG    Adding transaction 19
2022-09-12 13:19:49 DEBUG    Getting transaction 19
2022-09-12 13:19:49 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-09-12 13:19:49 WARNING  Modbus connection failed
Modbus connection failed
2022-09-12 13:19:49 DEBUG    Exception Response(131, 3, IllegalAddress)
2022-09-12 13:19:49 INFO     Scraping: 9/18 registers failed to scrape
2022-09-12 13:19:49 DEBUG    Using Inverter Time: 2022-9-12 10:19:46
2022-09-12 13:19:49 INFO     Inverter: Successfully scraped in 7.660853 secs
2022-09-12 13:19:49 DEBUG    PVOutput: Data Logged: {'v1': 3800.0, 'v2': 606, 'v4': 33373, 'v6': 219.0, 'count': 1}
2022-09-12 13:19:49 DEBUG    PVOutput: Request; https://pvoutput.org/service/r2/addbatchstatus.jsp, {'X-Pvoutput-Apikey': 'xxxxxxxxx', 'X-Pvoutput-SystemId': '93119', 'Content-Type': 'application/x-www-form-urlencoded', 'cache-control': 'no-cache'} : {'data': '20220912,10:19,3800,606,,33373,,219.0,,,,,,', 'c1': 2}
2022-09-12 13:19:49 DEBUG    Starting new HTTPS connection (1): pvoutput.org:443
2022-09-12 13:19:50 DEBUG    https://pvoutput.org:443 "POST /service/r2/addbatchstatus.jsp?data=20220912%2C10%3A19%2C3800%2C606%2C%2C33373%2C%2C219.0%2C%2C%2C%2C%2C%2C&c1=2 HTTP/1.1" 200 16
2022-09-12 13:19:50 INFO     PVOutput: Data uploaded
2022-09-12 13:19:50 INFO     Closing Session: ModbusTcpClient(192.168.1.208:502)
2022-09-12 13:19:50 DEBUG    Processing Time: 8.52 secs`

lexleite, 2022-09-15:
I saw that there was a work to make it work with sg5kd and sg8kd. Are those 2 similar in modbus? I forced it to think that im using sg5kd and it seems to be detecting stuff correctly. Maybe they share the same modbus registers?

michbeck100, 2022-09-15:
Would be interesting what device type code you see when running SunGather: https://github.com/bohdan-s/SunGather/blob/27bf7dcb90d7240539199e41aa335a8c6eb96c91/SunGather/registers-sungrow.yaml#L29
I would personally think that using SG5K-D should return proper results.

lexleite, 2022-09-15:
It shows "None" on the webpage

bohdan-s, 2022-09-15:
Key part,Unknown Type Code Detected: 290 Just need to add that in to the lookup table and see if I can dig up a offical list of supported registers. Otherwise I’ll just set to the “base” set all models seem to support. Regards,Bohdan

michbeck100, 2022-09-15:
That's obvious because the mapping for your inverter doesn't exist in https://github.com/bohdan-s/SunGather/blob/main/SunGather/registers-sungrow.yaml#L29. If you have auto detection on, it should log a warning `Unknown Type Code Detected`. Do you see this in the logs?

michbeck100, 2022-09-15:
Then please try to add 0x122 to the `datarange` and the correct value and see what happens:

```
        - response: 0x122
          value: "SG3K-D"
```

bohdan-s, 2022-09-15:
It will probably not read anything,Since that model has no registers listed. Regards,Bohdan

michbeck100, 2022-09-15:
Oh, yes. Then just search and replace `"SG5K-D",` with `"SG3K-D","SG5K-D",` 😉

lexleite, 2022-10-06:
I tried to create the elements for the 3kd with no success. I guess the best option is to copy 5kd to 3kd.

danpadure, 2022-10-24:
I just managed to do something similar for my SG3.0RS

sheepy971, 2026-10-18:
Hey. It seems my inverter isn’t supported yet. I’m getting the following error: code 3597. I have an SH6.0RT inverter. Thanks a lot for your help!