bohdan-s/SunGather #77: new install module not found

danpadure, 2022-10-21:
I can't understand if the problem is with SunGather (I think not) or with SungrowModbusTcpClient or pymodbus

In pymodbus/client there is no sync.py  file.  Only sync_diag.py

Traceback (most recent call last):
  File "/home/dan/SunGather/SunGather/sungather.py", line 3, in <module>
    from SungrowModbusTcpClient import SungrowModbusTcpClient
  File "/usr/local/lib/python3.9/dist-packages/SungrowModbusTcpClient/SungrowModbusTcpClient.py", line 1, in <module>
    from pymodbus.client.sync import ModbusTcpClient
ModuleNotFoundError: No module named 'pymodbus.client.sync'

bohdan-s, 2022-10-21:
Hi,There seems to be an issue where pymodbus v3 is not supported. In the requirements file force v2 (change >= to ==) and it will work. Will look at a PR to fix shortly. Regards,Bohdan

danpadure, 2022-10-22:
it worked
I get something, but only on http mode it's working ok.  In sungrow/modbus mode  I get some data but lots of modbus errors
I am accessing it through the WINET-S wifi dongle
Should I remove the dongle and connect the eth directly to the inverter in it's place?
It's a SG3.0RS  and it has code 9731 = 0x2603
I replaced SG3.0RT with the RS and I get some more information, but I'dd like also the MPPTs voltage/amps (has 2 MPPTs)

bohdan-s, 2022-10-22:
Correct, the WiNet-S does not fully work with modbus. You will need to use the HTTP method. In fact up until the last firmware update the WiNet-S had no modbus support, now it seems to have “some”. We are all hoping a future update fully resolves/enables modbus support. Regards,Bohdan

closed 2022-10-24 by danpadure