bohdan-s/SunGather #79: Hidden registers?

elektrinis, 2022-10-27:
Not really an issue, however I have two observations:
1. Looks like there are hidden registers, not provided in the documentation and not being used by SunGather
2. Most values just don't add up on my SH10RT.

1: As seen on screenshots below (iSolarCloud app), there are extra registers like `Total Backup Power` and per-phase `Backup` power. I can't find this data in official register map (the PDF), and in any of modbus implementations, including SunGather.
What I need the most is per-phase power, as measured by smart meter. Yes, my hybrid inverter, SH10RT, came with DTSU666 meter in the box. Also I want to note that per-phase power, reported by the app, is offset from real values (that I have measured with other equipment) by 1/3 of `Total Backup Power` (actual power on grid input is zero, although some phases are positive, some negative).

2: Another issue is that numbers don't add up. For example `battery power` is not equal to `battery current` and `battery voltage` multiplied. Same goes for other values, like `Total DC Power` does not equal the sum of load, grid and battery charge. I spent a couple of days trying to understand what is what, and still ended up asking about it here.
So far I came to conclusion that these values are being faked, because, for example, if battery power is equal load power, and everything else is disconnected, there is no leftover for internal conversion losses. And inverter runs pretty hot, there should be a significant loss.

Anyone looked in to this?

[image]
[image]
[image]
[image]

snaven1, 2022-12-03:
Hats off to the developer and the huge amount of work that has gone into the development of this interface. By and large it works extremely well. Following up on the above comment, yes also as an owner of a SH10RT, I have spent a huge amount of time mapping ISolarCloud parameters to the local web interface on the Sungrow unit and then onto the Sungather Modbus interface parameters. Additionally, I have crossed checked the modbus register information as per published documents from Sungrow and have found much of the information is simply not up to date and potentially wrong. As mentioned above, there simply is a whole bunch of registers that are not documented. The information from the three interfaces is simply inconsistent and hard to resolve. To that end, I have created a document that might help other parties that own a SH10RT, that at least attempts to align the various parameters across the interfaces and give some meaning to what the parameters mean. My work is not the bible on this issue, take it for its worth! 

[Sungrow Inverter Parameters.docx](https://github.com/bohdan-s/SunGather/files/10145817/Sungrow.Inverter.Parameters.docx)

snaven1, 2022-12-03:
Updated document

[Sungrow Inverter Parameters.docx](https://github.com/bohdan-s/SunGather/files/10145841/Sungrow.Inverter.Parameters.docx)

snaven1, 2023-02-16:
[Sungrow.Inverter.Parameters.docx](https://github.com/bohdan-s/SunGather/files/10752636/Sungrow.Inverter.Parameters.docx)