bohdan-s/SunGather #80: This nights clock change lead to freeze sungather

dhavlik, 2022-10-30:
Somehow sungather stopped working with the first scrape which took place after the clock change. See log:

`2022-10-30 02:59:54 INFO     Inverter: Successfully scraped in 4.950422 secs
2022-10-30 02:59:54 INFO     MQTT: Published
2022-10-30 02:59:54 INFO     Next scrape in 5 secs
2022-10-30 02:00:04 INFO     Inverter: Successfully scraped in 82805.131784 secs
2022-10-30 02:00:04 INFO     MQTT: Published
2022-10-30 02:00:04 WARNING  SunGather is taking 82805.13 to process, which is longer than interval 10, Please increase scan interval
SunGather is taking 82805.13 to process, which is longer than interval 10, Please increase scan interval`

There were no new log entries after the last one.

dhavlik, 2022-10-30:
See https://github.com/bohdan-s/SunGather/pull/81 for a fix.

michbeck100, 2022-12-26:
Can this be closed then?

dhavlik, 2022-12-26:
Yep. Thx.

closed 2022-12-26 by dhavlik