bohdan-s/SunGather #85: Sungrow SH10RT-112 'index out of range' error

dervomsee, 2022-11-13:
Hi,
i received last week my SHRT10-112 with the 9,6kWh Battery. The SHRT10-112 is a SH10RT with slightly modified hardware (or suppliers of hardware). [Differences and Compatibility of SHRT and SBR versions.pdf](https://github.com/bohdan-s/SunGather/files/9997424/Differences.and.Compatibility.of.SHRT.and.SBR.versions.pdf).

I'm using just ethernet connection, no WiFi connection. When connectin to the inverter with a modbus client and reading register 5000, i'm receiving 0xe0f. I extended already the registers-sungrow.yaml file to support the SHRT10-112, when receiving a 0xe0f. (see [registers-sungrow.yaml.txt](https://github.com/bohdan-s/SunGather/files/9997430/registers-sungrow.yaml.txt)).

The sungather.py connects and reads the values for the console output correct. But i get many 'index out of range' errors.
Here a console log of the first output: [debuglog.txt](https://github.com/bohdan-s/SunGather/files/9997434/debuglog.txt)

Is this normal? I necessary i can upload more/other debug outputs.

dervomsee, 2022-11-13:
With `connection: http` it works, partially... But many values are zero, e.g. daily_running_time=0 makes no sense.

```2022-11-13 15:31:52 INFO     Starting SunGather 0.3.8
2022-11-13 15:31:52 INFO     Loaded config: config-example.yaml
2022-11-13 15:31:53 INFO     Loaded registers: /mnt/c/Users/Cornelius/Downloads/SunGather/SunGather/registers-sungrow.yaml
2022-11-13 15:31:53 INFO     Registers file version: 0.2.0
+----------------------------------------------+
| Inverter Configuration Settings              |
+----------------------------------------------+
| Config             | Value                   |
+--------------------+-------------------------+
| host               | 192.168.20.12           |
| port               | 8082                    |
| timeout            | 10                      |
| retries            | 3                       |
| RetryOnEmpty       | False                   |
| slave              | 1                       |
| model              | SH10RT-112              |
| level              | 1                       |
| use_local_time     | False                   |
| smart_meter        | True                    |
| connection         | http                    |
+----------------------------------------------+
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| ----    | device_type_code                    | SH10RT-112           |
| vr003   | run_state                           | ON                   |
| 5003    | daily_power_yields                  | 6.5 kWh              |
| 5004    | total_power_yields                  | 237 kWh              |
| 5008    | internal_temperature                | 34.0 °C              |
| 5019    | phase_a_voltage                     | 229.6 V              |
| 5083    | meter_power                         | 0 W                  |
| 5091    | load_power                          | 546 W                |
| 5113    | daily_running_time                  | 0 min                |
| 6100    | pv_power_of_today                   | 0 W                  |
| 6196    | daily_pv_energy_yields              | 0.0 kWh              |
| 6290    | direct_power_consumption_today_pv   | 0 W                  |
| 6386    | direct_power_consumption_pv         | 0.0 kWh              |
| 6469    | export_power_from_pv_today          | 0 W                  |
| 6565    | export_power_from_pv                | 0.0 kWh              |
| 6648    | battery_charge_power_from_pv_today  | 0.0 kWh              |
| 6744    | battery_charge_power_from_pv        | 0.0 kWh              |
| 13003   | total_pv_generation                 | 27.4 kWh             |
| 13005   | daily_pv_export                     | 0.1 kWh              |
| 13006   | total_pv_export                     | 3.8 kWh              |
| 13008   | load_power_hybrid                   | 546 W                |
| 13010   | export_power_hybrid                 | 0 W                  |
| 13012   | daily_battery_charge_from_pv        | 4.2 kWh              |
| 13013   | total_battery_charge_from_pv        | 13.8 kWh             |
| 13017   | daily_direct_energy_consumption     | 3.8 kWh              |
| 13018   | total_direct_energy_consumption     | 9.8 kWh              |
| 13022   | battery_power                       | 404 W                |
| 13023   | battery_level                       | 43.1 %               |
| 13029   | self_consumption_of_day             | 97.4 %               |
| 13030   | grid_state                          | 0                    |
| 13034   | total_active_power                  | 546 W                |
| 13036   | daily_import_energy                 | 1.5 kWh              |
| 13037   | total_import_energy                 | 5.0 kWh              |
| 13045   | daily_export_energy                 | 0.1 kWh              |
| 5006    | start_stop                          | Start                |
| vr001   | export_to_grid                      | 0 W                  |
| vr002   | import_from_grid                    | 0 W                  |
| vr004   | timestamp                           | 2022-11-13 15:32:00  |
+----------------------------------------------------------------------+```

michbeck100, 2022-11-14:
Can you tell which values are missing? I would expect these to be the cause for the error. But without having an inverter to test it will be difficult to fix this. Could find any modbus documentation on the newer V112?

dervomsee, 2022-11-15:
Hmm, maybe most of it was a configuration error. 
When using 'connection: sungrow' and connecting to the integrated ethernet port (not the WiNet-S port) i get no errors.

Some values are zero, but they are for me not relevant. E.g. daily_running_time, pv_power_of_today, daily_pv_energy_yields, direct_power_consumption_today_pv, .. They are zero. But i can calculate them in influx too. 
[image]

With #87 this issue can be closed.

bohdan-s, 2022-11-16:
Merged #87

closed 2022-11-16 by bohdan-s