bohdan-s/SunGather #98: Feature Request: Export to Charge HQ

rianbk, 2023-01-07:
ChargeHQ allows you to send solar generation data to their API, this then helps smartly control charging of EV vehicles.

https://chargehq.net/kb/push-api

Now they do support Sungrow natively, but this uses the isolorcloud website and can have data delays of up to 10 minutes.
Considering sungather has access directly to the inverter and can easily do 30 second intervals, this would allow the EV charging to be more efficient.

bohdan-s, 2023-01-07:
Looks fairly easy to implement. 
If they support a free account I’ll have a look this week.

rianbk, 2023-01-07:
@bohdan-s ChargeHQ is currently free, so should be no issues there.

rianbk, 2023-04-27:
Hey @bohdan-s, my car is arriving in under a month so will be able to test when you have time for this one 😀