bohdan-s/SunGather #99: More info from the smart meter?

bows33r, 2023-01-11:
Is it possible to get out more info from the smart meter then just the power of the phases? 

I would like to get voltage and current also

michbeck100, 2023-01-12:
I thought this would be https://github.com/bohdan-s/SunGather/blob/main/SunGather/registers-sungrow.yaml#L299 onwards for voltage and https://github.com/bohdan-s/SunGather/blob/main/SunGather/registers-sungrow.yaml#L1345 onwards for current 🤔?

bows33r, 2023-01-12:
No they only shows a value when the solar panels is producing energy :(

michbeck100, 2023-01-12:
You are right for the current, but voltage works always:

[image]

bows33r, 2023-01-12:
hmm not for me :o

bohdan-s, 2023-01-12:
You can’t read the smart meter directly, only via the inverter. Some models of inverter don’t have all the registers available when they are not in run mode :(Sadly there isn’t much you can do unless you connect your own modbus hardware to the smart meter directly to read the values. Regards,Bohdan

s0laria, 2023-02-20:
In case someone is interested, the following registers hold power in watts for each phase. Only tested on SH10RT, so I don't know if it will work for other models:

    - name: "phase_a_power"
      level: 2
      address: 5603
      datatype: "S32"
      unit: "W"
      models: ["SH10RT"]

    - name: "phase_b_power"
      level: 2
      address: 5605
      datatype: "S32"
      unit: "W"
      models: ["SH10RT"]

    - name: "phase_c_power"
      level: 2
      address: 5607     
      datatype: "S32"
      unit: "W"
      models: ["SH10RT"]

michbeck100, 2023-02-20:
Found that, too. 

Additionally there is another one for the total power:

```yaml
- name: "meter_total_power"
  level: 2
  address: 5601
  datatype: "S32"
  unit: "W"
  models: ["SH5.0RS","SH3.6RS","SH4.6RS","SH6.0RS","SH10RT","SH10RT-V112","SH8.0RT","SH6.0RT","SH5.0RT"]
```

michbeck100, 2023-02-20:
And I can confirm that its working (at least on the SH10RT) with DTSU666.