bohdan-s/SunGather #101: Trouble to connect Mosquitto MQTT

lasse79, 2023-01-21:
Hi,
just got my SH10RT and try to implemt it to the home assistant.
Got SunGather running on eth port and getting out 39 registers. As soon i try to connect to the local Mosquitto MQTT Server SunGather quit with following faults:
`Traceback (most recent call last):
  File "./sungather.py", line 190, in <module>
    main()
  File "./sungather.py", line 158, in main
    export.publish(inverter)
  File "/root/SunGather/SunGather/exports/mqtt.py", line 123, in publish
    payload = json.dumps(inverter.inverter_config | inverter.client_config | inverter.latest_scrape).replace('"', '\"')
TypeError: unsupported operand type(s) for |: 'dict' and 'dict'
`
im running rasperryPi with python version 3.7.3

anybody lnow this issue or can support me?
Thanks in andvance!

michbeck100, 2023-01-21:
From the error message I would think that there's a problem with any of the 3 objects: `inverter.inverter_config | inverter.client_config | inverter.latest_scrape`. If you have the possibility to debug the code you could look into these objects. Or you just add this lines just before [line 123](https://github.com/bohdan-s/SunGather/blob/492b866fe34e87875dbbb2bc7cc24357d0f8d87e/SunGather/exports/mqtt.py#L123).

```python
logging.info(inverter.inverter_config)
logging.info(inverter.client_config)
logging.info(inverter.latest_scrape)
```

michbeck100, 2023-01-21:
Once you have the output please post it here.

lasse79, 2023-01-21:
here the logs:
`Logged 39 registers to Console
2023-01-21 18:36:02 INFO     MQTT: Published Home Assistant Discovery messages
2023-01-21 18:36:02 INFO     {'model': 'SH10RT', 'serial_number': 'A226106976', 'level': 1, 'scan_interval': 30, 'use_local_time': False, 'smart_meter': False, 'connection': 'modbus', 'slave': 1, 'start_time': ''}
2023-01-21 18:36:02 INFO     {'host': '192.168.100.33', 'port': 502, 'timeout': 10, 'retries': 3, 'RetryOnEmpty': False}
2023-01-21 18:36:02 INFO     {'device_type_code': 'SH10RT', 'run_state': 'ON', 'last_reset': '2023-1-21 18:36:00', 'daily_export_to_grid': 0.0, 'daily_import_from_grid': 0.0, 'daily_power_yields': 0.0, 'total_power_yields': 0, 'internal_temperature': 4.7, 'phase_a_voltage': 218.7, 'daily_running_time': 0, 'pv_power_of_today': 0, 'daily_pv_energy_yields': 0.0, 'direct_power_consumption_today_pv': 0, 'direct_power_consumption_pv': 0.0, 'export_power_from_pv_today': 0, 'export_power_from_pv': 0.0, 'battery_charge_power_from_pv_today': 0.0, 'battery_charge_power_from_pv': 0.0, 'total_pv_generation': 0.0, 'daily_pv_export': 0.0, 'total_pv_export': 0.0, 'load_power_hybrid': 0, 'export_power_hybrid': 0, 'daily_battery_charge_from_pv': 0.0, 'total_battery_charge_from_pv': 0.0, 'daily_direct_energy_consumption': 0.0, 'total_direct_energy_consumption': 0.0, 'battery_power': 0, 'battery_level': 11.5, 'self_consumption_of_day': 0.0, 'grid_state': None, 'total_active_power': 0, 'daily_import_energy': 0.0, 'total_import_energy': 0.0, 'daily_export_energy': 0.0, 'start_stop': 'Start', 'timestamp': '2023-1-21 18:36:00', 'export_to_grid': 0, 'import_from_grid': 0}

Traceback (most recent call last):
  File "./sungather.py", line 190, in <module>
    main()
  File "./sungather.py", line 158, in main
    export.publish(inverter)
  File "/root/SunGather/SunGather/exports/mqtt.py", line 126, in publish
    payload = json.dumps(inverter.inverter_config | inverter.client_config | inverter.latest_scrape).replace('"', '\"')
TypeError: unsupported operand type(s) for |: 'dict' and 'dict'

`

sorry my python skills are too bad to solve by my selfe, but anyway bayme it helps generall or other users as well

bohdan-s, 2023-01-21:
Can you try setting smart meter to true?

michbeck100, 2023-01-21:
Another guess would be to update your python version.

michbeck100, 2023-01-21:
I think `TypeError: unsupported operand type(s) for |: 'dict' and 'dict'` means that your python version doesn't support merging multiple json objects using the `|` operand.

lasse79, 2023-01-22:
On buisness trip, will look into this next weekend, thanks for support

lasse79, 2023-03-19:
sorry for the delay...
on my side it was a combination on a wrong Firmware on the inverter and the not connected smart meter in the beginning.
Right now it looks like it works and i can readout 39 registers!
Thanks for the support

pkerney, 2023-05-09:
I had same issue on Centos 8.5 until... as called out by @michbeck100 
# yum install python39
# pip3.9 install --upgrade -r requirements.txt
# python3.9 sungather.py

jpwise9, 2024-04-04:
Similarly had this issue - python 3.7 as part of openhabian.
Managed to cludge the following workaround to merge the fields instead of using the | merge operand that wasn't supported until later versions.

exports/mqtt.py - line 123
#        payload = json.dumps(inverter.inverter_config | inverter.client_config | inverter.latest_scrape).replace('"', '\"')
        payload_merge = {}
        payload_merge = inverter.inverter_config
        payload_merge.update(inverter.client_config)
        payload_merge.update(inverter.latest_scrape)
        payload = json.dumps(payload_merge).replace('"', '\"')

Not fully tested, and I don't use HA - so I don't now if the merged output is identical to the original format, but it was sufficient to get the mqtt export enabled and functioning.

Thx.