bohdan-s/SunGather #108: Docker (Compose) log file permissions error "Permission denied: '/logs/SunGather.log'"

iDontWantAUsername, 2023-03-02:
Hi,
I am running this in Docker-Compose:
```
version: '3.3'
volumes:
  sungather_logs:

services:
  sungather:
    container_name: sungather
    image: bohdans/sungather:latest
    network_mode: host
    ports:
      - 8090:8090
    volumes:
      - /etc/sungather/config:/config/
#      - /var/log/sungather/:/logs/
#      - /etc/sungather/logs:/logs/
      - sungather_logs:/logs/
    environment:
      - TZ=Australia/Melbourne
    restart: unless-stopped
```
I have tried all 3 of the volumes for the logging however in all cases i get:

```
sungather    | 2023-03-02 21:19:42 INFO     Starting SunGather 0.5.0
sungather    | 2023-03-02 21:19:42 INFO     Need Help? https://github.com/bohdan-s/SunGather
sungather    | 2023-03-02 21:19:42 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
sungather    | 2023-03-02 21:19:42 INFO     Loaded config: /config/config.yaml
sungather    | 2023-03-02 21:19:43 INFO     Loaded registers: registers-sungrow.yaml
sungather    | 2023-03-02 21:19:43 INFO     Registers file version: 0.2.2
sungather    | Traceback (most recent call last):
sungather    |   File "/usr/src/sungather/sungather.py", line 190, in <module>
sungather    |     main()
sungather    |   File "/usr/src/sungather/sungather.py", line 105, in main
sungather    |     fh = logging.handlers.RotatingFileHandler(logfile, mode='w', encoding='utf-8', maxBytes=10485760, backupCount=10) # Log 10mb files, 10 x files = 100mb
sungather    |          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sungather    |   File "/usr/local/lib/python3.11/logging/handlers.py", line 155, in __init__
sungather    |     BaseRotatingHandler.__init__(self, filename, mode, encoding=encoding,
sungather    |   File "/usr/local/lib/python3.11/logging/handlers.py", line 58, in __init__
sungather    |     logging.FileHandler.__init__(self, filename, mode=mode,
sungather    |   File "/usr/local/lib/python3.11/logging/__init__.py", line 1181, in __init__
sungather    |     StreamHandler.__init__(self, self._open())
sungather    |                                  ^^^^^^^^^^^^
sungather    |   File "/usr/local/lib/python3.11/logging/__init__.py", line 1213, in _open
sungather    |     return open_func(self.baseFilename, self.mode,
sungather    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sungather    | PermissionError: [Errno 13] Permission denied: '/logs/SunGather.log'

```
When the log file is active in the config e.g.: `log_file: DEBUG`

Is this a defect with the docker image?

bohdan-s, 2023-03-02:
Let me test and get back to you :)

Octofinger, 2023-03-29:
What permissions do you have on the directory sungather_logs? Where is the directory located?
You should use absolute paths on paths to volumes. Then check the permissions. You should have write permissions on the same user that starts the docker container.

dewenni, 2023-06-18:
I had the same problem.
Adding `user: root` fixed it for me

```
services:
  sungather:
    image: bohdans/sungather:latest
    container_name: sungather
    restart: always
    network_mode: host
    user: root
    environment:
      - TZ=Europe/Berlin
    ports:
      - "8080:8080"
    volumes:
      - /volume1/docker/sungather/config.yaml:/config/config.yaml
      - /volume1/docker/sungather/logs/:/logs/
```