bohdan-s/SunGather #111: SG8.0RS Index Out of Range error

iDontWantAUsername, 2023-03-03:
Hi, 
I know this inverter isn't official supported but it seems to mostly work however i do get an Index Out of Range error when returning all Level 3 - am using this to get the MPPT outputs to check the output from each array (and this isnt' returned with level 2 - probably due to the unsupported inverter.

Log file is:
```
Attaching to sungather
sungather    | 2023-03-03 19:07:22 INFO     Starting SunGather 0.5.0
sungather    | 2023-03-03 19:07:22 INFO     Need Help? https://github.com/bohdan-s/SunGather
sungather    | 2023-03-03 19:07:22 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
sungather    | 2023-03-03 19:07:22 INFO     Loaded config: /config/config.yaml
sungather    | 2023-03-03 19:07:23 INFO     Loaded registers: registers-sungrow.yaml
sungather    | 2023-03-03 19:07:23 INFO     Registers file version: 0.2.2
sungather    | 2023-03-03 19:07:23 INFO     Logging to console set to: DEBUG
sungather    | 2023-03-03 19:07:23 DEBUG    Inverter Config Loaded: {'host': '192.168.30.20', 'port': 502, 'timeout': 10, 'retries': 3, 'slave': 1, 'scan_interval': 30, 'connection': 'modbus', 'model': 'SG8.0RS', 'smart_meter': True, 'use_local_time': False, 'log_console': 'DEBUG', 'log_file': 'OFF', 'level': 3}
sungather    | 2023-03-03 19:07:23 INFO     Loading SungrowClient 0.1.0
sungather    | 2023-03-03 19:07:23 DEBUG    Checking Modbus Connection
sungather    | 2023-03-03 19:07:23 INFO     Modbus client is not connected, attempting to reconnect
sungather    | 2023-03-03 19:07:23 INFO     Connection: ModbusTcpClient(192.168.30.20:502)
sungather    | 2023-03-03 19:07:23 DEBUG    Connection to Modbus server established. Socket ('192.168.10.20', 34791)
sungather    | 2023-03-03 19:07:26 INFO     Bypassing Model Detection, Using config: SG8.0RS
sungather    | 2023-03-03 19:07:26 DEBUG    load_registers: read, 4989:10
sungather    | 2023-03-03 19:07:26 DEBUG    Current transaction state - IDLE
sungather    | 2023-03-03 19:07:26 DEBUG    Running transaction 1
sungather    | 2023-03-03 19:07:26 DEBUG    SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x7d 0x0 0xa
sungather    | 2023-03-03 19:07:26 DEBUG    New Transaction state 'SENDING'
sungather    | 2023-03-03 19:07:26 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
sungather    | 2023-03-03 19:07:26 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
sungather    | 2023-03-03 19:07:26 DEBUG    RECV: 0x0 0x1 0x0 0x0 0x0 0x17 0x1 0x4 0x14 0x41 0x32 0x32 0x43 0x30 0x37 0x30 0x32 0x35 0x31 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
sungather    | 2023-03-03 19:07:26 DEBUG    Processing: 0x0 0x1 0x0 0x0 0x0 0x17 0x1 0x4 0x14 0x41 0x32 0x32 0x43 0x30 0x37 0x30 0x32 0x35 0x31 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
sungather    | 2023-03-03 19:07:26 DEBUG    Factory Response[ReadInputRegistersResponse: 4]
sungather    | 2023-03-03 19:07:26 DEBUG    Adding transaction 1
sungather    | 2023-03-03 19:07:26 DEBUG    Getting transaction 1
sungather    | 2023-03-03 19:07:26 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
sungather    | 2023-03-03 19:07:26 INFO     Detected Serial: #######
sungather    | 2023-03-03 19:07:26 INFO     Closing Session: ModbusTcpClient(192.168.30.20:502)
sungather    | 2023-03-03 19:07:26 INFO     Loading Export: exports\webserver
sungather    | 2023-03-03 19:07:26 INFO     Webserver: Configured
sungather    | 2023-03-03 19:07:27 INFO     Loading Export: exports\mqtt
sungather    | 2023-03-03 19:07:27 DEBUG    Checking Modbus Connection
sungather    | 2023-03-03 19:07:27 INFO     Modbus, Connecting new session
sungather    | 2023-03-03 19:07:27 INFO     MQTT: Connected to 192.168.10.20:1883
sungather    | 2023-03-03 19:07:27 DEBUG    Connection to Modbus server established. Socket ('192.168.10.20', 37999)
sungather    | 2023-03-03 19:07:27 DEBUG    Scraping: read, 4949:50
sungather    | 2023-03-03 19:07:27 DEBUG    load_registers: read, 4949:50
sungather    | 2023-03-03 19:07:27 DEBUG    Current transaction state - TRANSACTION_COMPLETE
sungather    | 2023-03-03 19:07:27 DEBUG    Running transaction 2
sungather    | 2023-03-03 19:07:27 DEBUG    SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x55 0x0 0x32
sungather    | 2023-03-03 19:07:27 DEBUG    New Transaction state 'SENDING'
sungather    | 2023-03-03 19:07:27 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
sungather    | 2023-03-03 19:07:27 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
sungather    | 2023-03-03 19:07:27 DEBUG    RECV: 0x0 0x2 0x0 0x0 0x0 0x67 0x1 0x4 0x64 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x41 0x32 0x32 0x43 0x30 0x37 0x30 0x32 0x35 0x31 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
sungather    | 2023-03-03 19:07:27 DEBUG    Processing: 0x0 0x2 0x0 0x0 0x0 0x67 0x1 0x4 0x64 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x41 0x32 0x32 0x43 0x30 0x37 0x30 0x32 0x35 0x31 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
sungather    | 2023-03-03 19:07:27 DEBUG    Factory Response[ReadInputRegistersResponse: 4]
sungather    | 2023-03-03 19:07:27 DEBUG    Adding transaction 2
sungather    | 2023-03-03 19:07:27 DEBUG    Getting transaction 2
sungather    | 2023-03-03 19:07:27 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
sungather    | 2023-03-03 19:07:27 DEBUG    Scraping: read, 5000:38
sungather    | 2023-03-03 19:07:27 DEBUG    load_registers: read, 5000:38
sungather    | 2023-03-03 19:07:27 DEBUG    Current transaction state - TRANSACTION_COMPLETE
sungather    | 2023-03-03 19:07:27 DEBUG    Running transaction 3
sungather    | 2023-03-03 19:07:27 DEBUG    SEND: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x26
sungather    | 2023-03-03 19:07:27 DEBUG    New Transaction state 'SENDING'
sungather    | 2023-03-03 19:07:27 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
sungather    | 2023-03-03 19:07:27 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
sungather    | 2023-03-03 19:07:27 DEBUG    RECV: 0x0 0x3 0x0 0x0 0x0 0x4f 0x1 0x4 0x4c 0x0 0x50 0x0 0x0 0x1 0xa9 0x0 0x2b 0x0 0x0 0x0 0xd 0x0 0x0 0x1 0x62 0x3 0x7e 0x0 0x0 0x8 0xd2 0x0 0x4 0xb 0x9f 0x0 0x2 0x9 0x35 0x0 0x2 0x1 0x7 0x0 0x0 0x9 0x81 0x0 0x0 0x0 0x0 0x0 0x2c 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x3 0x0 0x0 0x3 0x58 0x0 0x0 0x1 0x0 0x1 0xf4 0x0 0x0 0x0 0x0
sungather    | 2023-03-03 19:07:27 DEBUG    Processing: 0x0 0x3 0x0 0x0 0x0 0x4f 0x1 0x4 0x4c 0x0 0x50 0x0 0x0 0x1 0xa9 0x0 0x2b 0x0 0x0 0x0 0xd 0x0 0x0 0x1 0x62 0x3 0x7e 0x0 0x0 0x8 0xd2 0x0 0x4 0xb 0x9f 0x0 0x2 0x9 0x35 0x0 0x2 0x1 0x7 0x0 0x0 0x9 0x81 0x0 0x0 0x0 0x0 0x0 0x2c 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x3 0x0 0x0 0x3 0x58 0x0 0x0 0x1 0x0 0x1 0xf4 0x0 0x0 0x0 0x0
sungather    | 2023-03-03 19:07:27 DEBUG    Factory Response[ReadInputRegistersResponse: 4]
sungather    | 2023-03-03 19:07:27 DEBUG    Adding transaction 3
sungather    | 2023-03-03 19:07:27 DEBUG    Getting transaction 3
sungather    | 2023-03-03 19:07:27 DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
[... 79 lines cut ...]
sungather    | pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
sungather    | Modbus Error: [Input/Output] Unable to decode request
sungather    | Traceback (most recent call last):
sungather    |   File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 208, in execute
sungather    |     self.client.framer.processIncomingPacket(response,
sungather    |   File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
sungather    |     self._process(callback)
sungather    |   File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
[... cut ...]
sungather    | 2023-03-03 19:07:27 DEBUG    Connection to Modbus server established. Socket ('192.168.10.20', 37771)
sungather    | 2023-03-03 19:07:27 DEBUG    Current transaction state - TRANSACTION_COMPLETE
sungather    | 2023-03-03 19:07:27 DEBUG    Running transaction 9
```

Is it possible to look into the Index error and/or add SG8.0RS as a supported inverted? Let me know what information you need if this is to become supported.

I am using Docker-Compose if that make a difference to anything.

Thanks!

bohdan-s, 2023-03-03:
Hi,
LVL3 scans EVERY register. So out of bounds are expected. 
You could ask Sungrow support for the modbus document for your register and I can update the official. 
You could also let me know how many MPPT / Strings your model supports and I can add support at LVL 2 as it should be. 
lvl 1 = standard registers, lvl 2 = all your model supports (when added). Lvl3 = all known registers, supported or not.

iDontWantAUsername, 2023-03-03:
Hi,
That makes sense.

The SG8.0RS supports 3 Strings.

I will drop SunGrow an email to get the full supported modbus registers.

Thanks,
Richard

iDontWantAUsername, 2023-03-03:
It looks like the issue could be that SG8.0RS has been added to the `string_3_voltage`, `string_3_current` registers when in-fact it outputs to `mppt_3_voltage`, `mppt_3_current`.

I have sent SunGrow a message so will update if they return a full list

bohdan-s, 2023-03-03:
MPPT and String are 2 different things. 
for example I have 2 MPPT, 3;3;2 strings for a total of 8 strings. 
The PV document had all the models and how many they have. Hybrid models don’t seem to use MPPT just strings.

iDontWantAUsername, 2023-03-03:
Ahr ok, so it just looks like the SG#.0RS Inverters may be missing their MPPT_3 resources, these have 3 MPPT and 1 default string per MPPT.

iDontWantAUsername, 2023-03-07:
Latest ModBus mappings attached, it looks like it is a few years old - it does have the a list of how many MPPT channels each inverter has in the appendix 5. I know it is a big document - would you like me to extract any information from it to help?

[TI_20211225_Communication Protocol of PV Grid-Connected String Inverters_V1.1.40_EN.pdf](https://github.com/bohdan-s/SunGather/files/10908154/TI_20211225_Communication.Protocol.of.PV.Grid-Connected.String.Inverters_V1.1.40_EN.pdf)