bohdan-s/SunGather #115: Array registers incorrectly interpreted as single value

Octofinger, 2023-03-21:
There are some array registers in the read area that are configured as single value (U16) values although they are array values.
Eg:
"pv_power_of_today" is defined as one U16, but is rather an array of 96 U16 values showing the PV power throughout the day. One U16 for each 15 minute time frame. Not sure if it's a sliding window or if it resets at midnight and starts over.
"daily_pv_energy_yields" is defined as one U16 but is an array of 31 U16 that show the PV energy yield per day in current month.

Other fields exhibiting the same behavior are 
- monthly_pv_energy_yields
- yearly_pv_energy_yields
- direct_power_consumption_today_pv
- direct_power_consumption_pv
- direct_power_consumption_monthly_pv
- direct_power_consumption_yearly_pv
- export_power_from_pv_today
- export_power_from_pv
- export_power_from_pv_monthly
- export_power_from_pv_yearly
- battery_charge_power_from_pv_today
- battery_charge_power_from_pv
- battery_charge_power_from_pv_monthly
- battery_charge_power_from_pv_yearly

I believe we need a new datatype to handle these fields in the registers yaml to show that this is an array that starts from a certain address, has a certain length and each element has a specific datatype.
Maybe another array: scalar that adds array size to a register definition? If not present, the register is a single value.

I'm just not sure what implications this has on the scan and registry match, and the integrations using SunGather. But in it's current implementation, these registers return incomplete and incorrect data.

bohdan-s, 2023-03-21:
I am already aware of this and working on a way to read it :)
Though if polling every 30 secs you don’t need all 15 min history ;)

Octofinger, 2023-03-22:
I don't think they should be scraped by the recurring loop. The recurring loop gathers all momentary values and leaves for the receiving part to keep history. In the case of these arrays with history, they should rather be called explicitly by something that needs history. So I believe these registers should be treated differently, perhaps add a register level 4 that has to be explicitly set to scrape these arrays? Maybe a different function call?