bohdan-s/SunGather #118: Sungrow SH10RT(V112) with 11 byte serials

florianbrede-ayet, 2023-04-20:
I have new SH10RT from around 1/23 and it looks like Sungrow has expanded their serials to 11 bytes.

```
Devices: 2 x SH10RTV112
Serial A: A12B1234561
Serial B: A12B1234562
```

Unfortunately, `SunGather` or rather the `SungrowClient` only processes the first 10 bytes of the serial so both are identified as `A12B123456`.

vetinari, 2024-05-11:
What you get  back  is basically a `\0` terminated string - so concatenating all registers from 4990 (to 4999) as bytes, cutting from the first 0 byte to the end gives you a variable length string.

vetinari, 2024-05-11:
something like this should work
```python
                   elif register.get('datatype') == "UTF-8": # This seems to be Serial only, 10 uint16
                        utf_value = register_value.to_bytes(2, 'big')
                        for x in range(1,10):
                            utf_value += rr.registers[num+x].to_bytes(2, 'big')
                        register_value = utf_value.decode()
                        z = register_value.find('\0')
                        if z != -1:
                            register_value = register_value[:z]
```