bohdan-s/SunGather #119: Several inverters installed, only receiving data from one inverter

ui67, 2023-04-28:
We have installed two SG50CX inverters. Both can be accessed via the same IP. However, Sungather only outputs data from one inverter. The data from the second inverter is not being transmitted. I have already tried to create a second Sungather instance and switch the serial number to the second inverter in the config.yaml. Unfortunately, Sungather still reads the data from the other inverter. Is there a way to read both inverters simultaneously or add the data from both inverters so that it can be directly integrated into Home Assistant? I would greatly appreciate your help.

bohdan-s, 2023-04-28:
The serial number is for uniquely identifying the inverter after it’s read. it won’t impact how it reads data. 
I’m not sure how the 2 may be connected, but you could change the slave ID and see if the inverters are set up that way.

ui67, 2023-05-03:
I changed the the slave ID. It is working now. I am running two sungather instances now, and added them into Grafana.

closed 2023-05-03 by ui67