bohdan-s/SunGather #120: Openhab, modbus-binding and Sungather

olili, 2023-05-01:
First of all - many thanks for this great SunGather project.  
Really easy to connect Sungrow inverters to other devices over WiNet-S and mqtt.  

Nevertheless I would like to connect directly openhab via a regular bidirectional modbus-binding to the Inverter. 
(currently I do it oneway via Sungather --> Mosquitto --> Openhab-Mqtt-Binding).

Is there any "modbus-Proxy" ("regular modbus" to "http/winet-S-modibus" ) available? 

Best regards, O.

michbeck100, 2023-05-01:
You can use https://pypi.org/project/modbus-proxy/. There’s also a docker image at https://hub.docker.com/r/tiagocoutinho/modbus-proxy. You just have to point the proxy to your inverter and sungather, OpenHab etc. to the proxy then.

olili, 2023-05-01:
thx. This is exactly what I was looking for.
But is it somehow necessary to include SungrowModbusWebClient as I have only a Winet-S connection to the inverter. 
If yes, how?

bohdan-s, 2023-05-01:
The webclient part translates http to modbus for dongles that don’t support modbus.

olili, 2023-05-01:
thx. i know.
And I'm looking for an possibility to bridge/proxy to winet-s-http-"modbus"
to regular tcp modbus.
Any hint?