bohdan-s/SunGather #123: Scan level in config.yaml - nothing above level 1 seems to work

bleughb, 2023-05-11:
Hi,
Trying to get MPPT info out of my inverter (SG50RS) - but editing the config.yaml to scan at level 2 doesn't seem to work.

if I check Device MPPT log, I see only Level 1 - as below....
QoS: 0
Payload: {"model": "SG5.0RS", "serial_number": "DELETED", "level": 1, "scan_interval": 30, "use_local_time": false, "smart_meter": true, "connection": "modbus", "slave": 1, "start_time": "", "host": "192.168.0.25", "port": 502, "timeout": 10, "retries": 3, "RetryOnEmpty": false, "device_type_code": "SG5.0RS", "run_state": "ON", "last_reset": "2023-3-9 19:47:51", "daily_export_to_grid": 0.0, "daily_import_from_grid": 0.0,

PBanDee, 2023-08-23:
I have the same issue with my SG8.0RT. No effect changing level 2 or 3, only level 1 values are visible. My inverter connected to local network via UTP through WiNet-S module.

kosacke, 2024-04-21:
Hello,

sorry for posting to this old issue, but I have the same problem and since there is no solution posted here yet... 

SH10RT inverter
Cabled Modbus connection via the inverters LAN/Ethernet port (not Winet-s) 

I tried to read more registers than Level 1,but my setting in the config.yaml file is ignored. Output still shows the same number of registers and also "Level 1".

This is my config.yaml file:

```
host: 192.168.178.44
timeout: 10
scan_interval: 30
connection: modbus
smart_meter: true
custom_mqtt_server: false
log_console: INFO
Level: 3

```

Where can I find and check the registers-sungrow.yaml file? With standard HA File Editor I can only open files in homeassistant folder.

Any help is greatly appreciated.

Johannes

bohdan-s, 2024-04-21:
Yaml config is case sensitive and also indentation matters, also make sure there is no # at the beginning of the line as anything after a # is ignored. 
Could you post the config snippet if it’s still not working?

kosacke, 2024-04-21:
I tried both Level and level, as this is not quite clear in the documentation. None of both works.

[image]

SunGather Frontend reads like this (sorry I have to post as screenshot, since copying from HA also copies the HTML formatting 
[image]

This is the log after starting SunGather:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-04-22 00:25:17 INFO     Starting SunGather 0.5.2
2024-04-22 00:25:17 INFO     Need Help? https://github.com/bohdan-s/SunGather
2024-04-22 00:25:17 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2024-04-22 00:25:17 INFO     Loaded config: /share/SunGather/config.yaml
2024-04-22 00:25:19 INFO     Loaded registers: registers-sungrow.yaml
2024-04-22 00:25:19 INFO     Registers file version: 0.2.4
2024-04-22 00:25:19 INFO     Logging to console set to: INFO
2024-04-22 00:25:19 INFO     Loading SungrowClient 0.1.0
2024-04-22 00:25:19 INFO     Modbus client is not connected, attempting to reconnect
2024-04-22 00:25:19 INFO     Connection: ModbusTcpClient(192.168.178.44:502)
2024-04-22 00:25:22 INFO     Detected Model: SH10RT-V112
2024-04-22 00:25:22 INFO     Detected Serial: A234170252
2024-04-22 00:25:22 INFO     Closing Session: ModbusTcpClient(192.168.178.44:502)
2024-04-22 00:25:22 INFO     Loading Export: exports\webserver
2024-04-22 00:25:22 INFO     Webserver: Configured
2024-04-22 00:25:22 INFO     Loading Export: exports\mqtt
2024-04-22 00:25:22 INFO     Modbus, Connecting new session
2024-04-22 00:25:22 INFO     Setting Initial Daily registers; daily_export_to_grid, daily_import_from_grid, last_reset
2024-04-22 00:25:22 INFO     Inverter: Successfully scraped in 0.72252 secs
2024-04-22 00:25:22 WARNING  MQTT: Server Disconnected; 0 messages queued, will automatically attempt to reconnect
MQTT: Server Disconnected; 0 messages queued, will automatically attempt to reconnect
2024-04-22 00:25:22 INFO     MQTT: Published Home Assistant Discovery messages
2024-04-22 00:25:22 INFO     MQTT: Registers Published
2024-04-22 00:25:22 INFO     Closing Session: ModbusTcpClient(192.168.178.44:502)
2024-04-22 00:25:22 INFO     Next scrape in 29 secs
2024-04-22 00:25:23 INFO     MQTT: Connected to core-mosquitto:1883

```

bohdan-s, 2024-04-21:
Interesting. 
let me play around on my end and find the issue.

kosacke, 2024-04-22:
Thank you for looking into it.
Other changes in the config.yaml are actually working, longer scan interval for example.

samuelkax, 2024-06-20:
I'm having the exakt same issue here. Connected via Wi-Fi with Winet-dongle. Using the home assistant addon.

Changing level in the config doesn't affect the output. Neither do the use_local_time option.

This is my config:
> host: 192.168.22.26
> timeout: 10
> scan_interval: 10
> connection: modbus
> smart_meter: false
> custom_mqtt_server: false
> log_console: INFO
> level: 3
> use_local_time: true

When I check the web ui, this is what I see (I have restarted the addon after changing the config):
[image]

However, if I install SunGather via docker on another machine it works. Then I can change the level and get all the data.

kosacke, 2024-10-22:
Hi everyone. Since there has been no update here for long, I abandoned the Addon for a while.

Now tried re-installing it but the error persists.

Is there any solution for this yet @bohdan-s?

I noticed that in the repository the Addon says version 0.1.4, see screenshot. According to github the current version is 0.5.2 (!). Any explanation for that? Can this be a cause to the level error? 
[image]

michbeck100, 2024-10-23:
The add-on version and Sungather differ because it's different projects (https://github.com/bohdan-s/hassio-repository/tree/main/SunGather vs. https://github.com/bohdan-s/SunGather). So that's normal.

kosacke, 2025-04-16:
No news here and I did not manage to get it working myself. Actually I am looking for other ways now to read Modbus Info from the inverter. But while doing this I looked into Supervisor protocols and found the following:

`2025-04-16 18:17:12.663 WARNING (MainThread) [supervisor.addons.options] Option 'level' does not exist in the schema for SunGather (7b536ee6_sungather)
`

Maybe that helps to identify what is causing the problem?