bohdan-s/SunGather #126: Support for multiple inverters?

erikderzweite, 2023-07-04:
Hi!

anyone had a success with more than one inverter connected together? I cannot scrape the modbus of the second inverter connected via LAN cable.

asdnjqiuwe, 2023-07-08:
I'm wondering this as well. I have two Inverters. The configuration seems to only support one config.yml, would you simply run two Docker instances where each of them point to a different config?

Not sure what you mean by "connected together" though, so perhaps the use case is different.

asdnjqiuwe, 2023-07-08:
That's good to hear, thanks!

Why are they connected together master-slave over modbus? I don't think my Inverters know about each other at all.

asdnjqiuwe, 2023-07-08:
Ok so the master-slave is a configuration in Sungrow, as far as I understand it only applies to Hybrid variants?

erikderzweite, 2023-07-10:
I don't know if other variants support the master-slave. I know though that one of the reasons for using it is German law requiring the export power to be limited to 70% of the installed power.
Master-Slave connection ensures that this threshold is not exceeded. Not sure about the reasons to use it outside Germany though as the power and load value of both systems cannot be read over modbus from master only.

jeffwood-lab, 2024-07-21:
> I've sucessfully started two instances after all. "Connected together" means in my case master-slave connection over modbus. You have to put slave = 0x02 for the second inverter though.

I do have the same problem. I have 3 inverters(each on one phase). I can only connect to one inverter at a time. I am using this as an add-on to HAOS. Have you been able to make this work on HA? Anyone else could come up with a solution?

julianbraun, 2024-11-10:
I tried to use I have two identical inverters and am currently running two Docker containers in parallel.
I also get all the data in HA, but here I have the problem that the name of the inverters is the model name, which is the same.
Not very nice and I can't manage to change the names at the moment.

julianbraun, 2024-11-10:
I have found a workaround:
- Create a Docker container for each inverter

If you have the same inverter multiple or want to change the Name in HA:
- Copy the file “registers-sungrow.yaml” in the volume “conf”
- Find out what the inverter designation in “registers...” is
- In “config.ymal”, set the model to the desired name
- Change the actual inverter designation in “registers-sungrow.yaml” to the desired name. (for all)
- If necessary, delete the MQTT integration and create a new one.