bohdan-s/SunGather #127: Missing data on WiNet-S dongle

selop, 2023-07-07:
Can anybody confirm that using a WiNet-S dongle /w modbus on a SH10RT-V112 inverter lacks some of the state registers:

```
| 13000   | system_state                        | Unknown              |
| 13001   | running_state                       | 0                    |
| 13001   | state_power_generated_from_pv       | 0                    |
| 13001   | state_battery_charging              | 0                    |
| 13001   | state_battery_discharging           | 0                    |
| 13001   | state_load_active                   | 0                    |
| 13001   | state_feed_into_grid                | 0                    |
| 13001   | state_import_from_grid              | 0                    |
| 13001   | state_power_generated_from_load     | 0
```

Especially the state of battery charge or discharge bit is important to me. 

My config.yaml:

```
inverter:
  host: 192.168.xx.xx                       # [Required] IP Address of the Inverter or Dongle
  slave: 0x01                               # [Optional] Default is 0x01
  scan_interval: 10                         # [Optional] Default is 30
  connection: modbus                        # [Required] options: modbus, sungrow, http
  smart_meter: True                         # [Optional] Default is False, Set to true if inverter supports reading grind / house consumption
  log_console: INFO                         # [Optional] Default is WARNING, Options: DEBUG, INFO, WARNING, ERROR
  level: 3                                  # [Optional] Set the amount of information to gather
                                            # 0 = Model and Solar Generation,
                                            # 1 (default) = Useful data, all required for exports,
                                            # 2 everything your Inverter supports,
                                            # 3 Everything from every register
```

DerDudeESP, 2023-07-07:
Yes same problem here.

btilson, 2023-07-26:
Same for me with the WiNet-S dongle connected to a SG8.0RS inverter.  I believe the WiNet-S dongle only exposes a subset of modbus registers.  I've found conflicting reports that Sungrow were going to release a firmware update for the WiNet-S dongle to resolve this but can't find anything concrete this is happening.

bohdan-s, 2023-07-26:
The issue actually seems to be the registers file. They are all trying to read 13001. 
I'lol have to go through the documentation and fix.

btilson, 2023-07-26:
> The issue actually seems to be the registers file. They are all trying to read 13001. I'lol have to go through the documentation and fix.

Hey mate, so are there equivalent registers to the 13xxx ones that are exposed via the WiNet-S dongle?

bohdan-s, 2023-07-26:
SunGather/registers-sungrow.yaml
there is a bunch of 13001 register entries. There should only be 1. That’s probably the cause.

Octofinger, 2023-07-30:
It's not an error. The entries each have a mask to retrieve the specific bit from running state. So there should be an entry per unique mask.
I can confirm that I can read the entries from my SH10RT-V112 using a WiNet-S over Wifi. I use http method in Sungather.
```
| 13000   | system_state                        | Forced Run           |
| 13001   | running_state                       | 41                   |
| 13001   | state_power_generated_from_pv       | 1                    |
| 13001   | state_battery_charging              | 0                    |
| 13001   | state_battery_discharging           | 0                    |
| 13001   | state_load_active                   | 1                    |
| 13001   | state_feed_into_grid                | 0                    |
| 13001   | state_import_from_grid              | 1                    |
| 13001   | state_power_generated_from_load     | 0                    |
```

selop, 2023-08-23:
@Octofinger This issue was filed with using modbus as the communication method. Using it with SH10RT-V112 and WiNet-S still does not return individual bits for the running state.

selop, 2023-08-28:
I agree it is not a Sungather problem but is rooted in the modbus implementation of the WiNet-S. I wanted to use the power of the community here to check whether Sungrow has updated the WiNet-S. Connecting the ethernet port of the device is deprecated according to Sungrow itself.

michbeck100, 2023-08-28:
Where did Sungrow say that the ethernet port is deprecated? Since I am using the old WiNet stick, which doesn't support real time data, this would be a giant drawback.

pohly, 2023-08-28:
> Where did Sungrow say that the ethernet port is deprecated?

Not sure whether it's any indicator, but at least the SG10RT and SG5RT that I just got don't have such a separate port.

btilson, 2023-08-28:
I have a SG8.0RS inverter and it also has no ethernet port, only the RJ-45 port for the WiNet-S dongle to connect to.  There must still be modbus implementation limitations on the WiNet-S dongle.  I have the latest firmware on it and get zero values for registers like daily import and daily export as just a couple of examples when using SunGather.  Currently I'm using Home Assistant to calculate these but that's not perfect as the values tend to drift a bit from what the iCloudSolar app and the inverter web interface report.

bohdan-s, 2023-08-28:
You also need a smart meter for any registers / values like import and export.

btilson, 2023-08-28:
Yeah I have a sungrow smart meter connected to my inverter and can see other meter related values e.g:

5083	meter_power	2905 W