bohdan-s/SunGather #133: How to discriminate between charging and discharching the battery? / Glitch in decimal

Petra-Kate, 2023-08-04:
Hello @all,

i)
scanning the output for my SunGrow SG10.0RT V112, I do not see a clue how to discriminate between charging and discharging the battery.

Any pointers, please?

ii)
Additionally, there seems to be a glitch in a decimal:

5036    | grid_frequency                      | 500.2 Hz

Cheers,
Petra

Octofinger, 2023-08-08:
Hi. Have a look at register 13001. It's a status register with several status bits. Those are mapped in the current registry file. More specifically, you want the registers ``state_battery_charging`` and ``state_battery_discharging``. Add them to your config file so they are mapped to HA and ensure you have ``level: 2`` to scan them.

On your second question, I can't reproduce it. I have a SH10RT, but the documentation says the register contains 0.1W precision. I get proper readings:
```
| 5036    | grid_frequency                      | 50.0 Hz              |
```

Petra-Kate, 2023-08-08:
The clue to get correct readings seems to lie in the choice of the inverter connection type: using "modbus" as I did initially reported the wrong values, as stated, and no 13001 register contents, even on level: 2. So did a change to "sungrow". Only the "http" protocol yields sensible values both for questions i) and ii).
So, for the time being, I consider the case as closed. Thanks a lot!

closed 2023-08-08 by Petra-Kate

dmindfuse, 2024-05-15:
Hello. 

I might add that this issue persists today, in my case for a SH8.0RT, whereby modbus reports a -1 persistently on both state_battery_charging and state_battery_discharging. Originally raised this question on michbeck's project, including screenshot, who kindly directed me here.
https://github.com/michbeck100/pv-monitoring/issues/14#issue-2297054593 

I suspect, but I don't know to debug/fix it, the registers are not configured correctly using modbus. I see both carry the same 13001 register number, which I suspect might be the issue.

I deployed the latest version of sungather this week, so I'm as up to date as possible.

Anyone?
Thanks.

dmindfuse, 2024-05-15:
Hi @michbeck100 , @bohdan-s ...

Considering the above... in my case the 0 -> -1 is problematic as 0 may have been "charging".  Strange, when I was running sungather configured with HTTP (versus now modbus) connectivity the issue was not present. 

Is SH8.0RT then the only inverter with a problem and can it be solved if I add 2 new registers for the same address - but it's not "pretty" -:

    - name: "state_battery_is_charging"
      level: 2
      address: 13001
      datatype: "U16"
      mask: 2
      datarange:
        - response: 1
          value: 1
        - response: 0
          value: 1
        - response: -1
          value: -1
      models: ["SH8.0RT","SH8.0RT-V112"]
    - name: "state_battery_is_discharging"
      level: 2
      address: 13001
      datatype: "U16"
      mask: 4
      datarange:
        - response: 1
          value: 1
        - response: 0
          value: 1
        - response: -1
          value: -1
      models: [""SH8.0RT","SH8.0RT-V112"]

And then remove SH8.0RT from the original registers.
Thoughts?