bohdan-s/SunGather #138: index out of range several sensors not updating

bqzero86, 2023-09-16:
Hi there,
struggeling with the initial setup to get all registers (standard ones) working. log is showing error with index out of range.

`2023-09-16 11:20:18 INFO     Starting SunGather 0.5.1
2023-09-16 11:20:18 INFO     Need Help? https://github.com/bohdan-s/SunGather
2023-09-16 11:20:18 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2023-09-16 11:20:18 INFO     Loaded config: /share/SunGather/config.yaml
2023-09-16 11:20:19 INFO     Loaded registers: registers-sungrow.yaml
2023-09-16 11:20:19 INFO     Registers file version: 0.2.3
2023-09-16 11:20:19 INFO     Logging to console set to: INFO
2023-09-16 11:20:19 INFO     Loading SungrowClient 0.1.0
2023-09-16 11:20:19 INFO     Modbus client is not connected, attempting to reconnect
2023-09-16 11:20:19 INFO     Connection: ModbusTcpClient(192.168.1.100:502)
2023-09-16 11:20:22 INFO     Detected Model: SH10RT-V112
2023-09-16 11:20:22 INFO     Detected Serial: A228261503
2023-09-16 11:20:22 INFO     Closing Session: ModbusTcpClient(192.168.1.100:502)
2023-09-16 11:20:22 INFO     Loading Export: exports\console
2023-09-16 11:20:22 INFO     Loading Export: exports\webserver
2023-09-16 11:20:22 INFO     Webserver: Configured
2023-09-16 11:20:22 INFO     Loading Export: exports\mqtt
2023-09-16 11:20:22 INFO     Modbus, Connecting new session
2023-09-16 11:20:22 INFO     MQTT: Connected to core-mosquitto:1883
2023-09-16 11:20:22 ERROR    index out of range
index out of range
2023-09-16 11:20:22 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-09-16 11:20:22 WARNING  Modbus connection failed
Modbus connection failed
2023-09-16 11:20:22 ERROR    index out of range
index out of range
2023-09-16 11:20:22 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-09-16 11:20:22 WARNING  Modbus connection failed
Modbus connection failed`

tried different approaches already. used dongle and modbus network port, makes no difference. Unable to setup the PV which is consisting of a SH10RT and SG6.0RT and SBR-128. I do get a an MQTT device named with my hybrid inverter, but not all registers show up as sensors. battery for example completely missing. Any idea how to proceed and make it fly? did not find a solution in the net.

br,
chris

vetinari, 2023-11-12:
SH5.0RT-V112 with the WiNet-S dongle here with similar issue: if you run it with DEBUG level, you probably see something like 

```
2023-11-06 21:34:05 DEBUG    Scraping: read, 5039:61
2023-11-06 21:34:05 DEBUG    load_registers: read, 5039:61
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 34613)
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 53395)
2023-11-06 21:34:05 DEBUG    Current transaction state - TRANSACTION_COMPLETE
2023-11-06 21:34:05 DEBUG    Running transaction 4
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 37443)
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 50989)
2023-11-06 21:34:05 DEBUG    SEND: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0xaf 0x0 0x3d
2023-11-06 21:34:05 DEBUG    New Transaction state 'SENDING'
2023-11-06 21:34:05 DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2023-11-06 21:34:05 DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2023-11-06 21:34:05 DEBUG    RECV: 0x0 0x4 0x0 0x0 0x0 0x2 0x1 0x84 0x2
2023-11-06 21:34:05 DEBUG    Processing: 0x0 0x4 0x0 0x0 0x0 0x2 0x1 0x84 0x2
2023-11-06 21:34:05 DEBUG    Factory Response[132]
2023-11-06 21:34:05 ERROR    index out of range
2023-11-06 21:34:05 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/var/lib/solar/.pyenv/versions/3.11.2/lib/python3.11/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/var/lib/solar/.pyenv/versions/3.11.2/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/var/lib/solar/.pyenv/versions/3.11.2/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
```

I think the `RECV:`line shows the full payload received, which according to [wikipedia](https://en.wikipedia.org/wiki/Modbus#Modbus_TCP) is broken down to

* 0x0 0x4 => transaction id (4, matches the above log)
* 0x0 0x0 => protocol version (always 0 for this)
* 0x0 0x2 => (remaining) length of the frame (2 bytes)
* 0x1 => unit id (i.e. the slave id, default is 0x1 in sungather)
* 0x84 => (132 in decimal) function code 
* 0x2 => where does this come from? the size was set to 2, not 3

vetinari, 2023-11-12:
Note: for every failing one, it's **always** the same payload (with a correct transaction id set) received, i.e. the last 7 bytes are the same, only the first two bytes with the transaction id change.

vetinari, 2023-11-12:
@bohdan-s why there's multiple connections open? ->
```
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 34613)
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 53395)
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 37443)
2023-11-06 21:34:05 DEBUG    Connection to Modbus server established. Socket ('192.168.8.10', 50989)
```

(from my log above) There's always a new local port, reusing the connection and just reconnect in case of errors should be fine.

bohdan-s, 2023-11-14:
This is fixed with new pymodbus dependency which required some rewrites to code to support. 

Sorry I have a backlog of issue fixes to push.