bohdan-s/SunGather #141: KeyError 'year'

Cuupa, 2023-10-28:
**Describe the bug**
SunGather is running inside a docker container. 

After some time the container exits with:

```
2023-10-28T07:55:00.606489156Z KeyError: 'year'
2023-10-28T07:55:00.606540032Z 
2023-10-28T07:55:00.606578406Z During handling of the above exception, another exception occurred:
2023-10-28T07:55:00.606635319Z 
2023-10-28T07:55:00.606673901Z Traceback (most recent call last):
2023-10-28T07:55:00.606743907Z   File "/usr/src/sungather/sungather.py", line 190, in <module>
2023-10-28T07:55:00.608404119Z     main()
2023-10-28T07:55:00.615251529Z   File "/usr/src/sungather/sungather.py", line 154, in main
2023-10-28T07:55:00.615465380Z     success = inverter.scrape()
2023-10-28T07:55:00.620974941Z               ^^^^^^^^^^^^^^^^^
2023-10-28T07:55:00.621090092Z   File "/usr/local/lib/python3.11/site-packages/SungrowClient/SungrowClient.py", line 405, in scrape
2023-10-28T07:55:00.624766765Z     del self.latest_scrape["year"]
2023-10-28T07:55:00.625040543Z         ~~~~~~~~~~~~~~~~~~^^^^^^^^
2023-10-28T07:55:00.625108749Z KeyError: 'year'
```

My docker-compose.yml

``` yaml
version: '3.8'

services:
  grafana:
    image: grafana/grafana:latest
    restart: unless-stopped
    depends_on:
      - influxdb-solar
    ports:
      - '13005:3000'
    volumes:
      - /volume2/docker/PRODUCTION/energy-monitoring/grafana/var/lib/grafana:/var/lib/grafana
      #- /volume2/docker/PRODUCTION/energy-monitoring/grafana/etc/grafana/provisioning:/etc/grafana/provisioning
      - /volume2/docker/PRODUCTION/energy-monitoring/grafana/etc/grafana/grafana.ini:/etc/grafana/grafana.ini:ro
    environment:
      PUID: 1060
      PGID: 100
  
  influxdb-solar:
    image: influxdb:latest
    restart: unless-stopped
    ports:
      - '18086:8086'
    volumes:
      - /volume2/docker/PRODUCTION/energy-monitoring/influxdb-solar/var/lib/influxdb2:/var/lib/influxdb2
    environment:
      PUID: 1060
      PGID: 100
      INFLUX_DB: <redacted>
      INFLUXDB_ADMIN_USER: <redacted>
      INFLUXDB_ADMIN_PASSWORD: <redacted>

  sungather:
    image: bohdans/sungather
    restart: unless-stopped
    ports:
      - '18087:8080'
    volumes:
      - /volume2/docker/PRODUCTION/energy-monitoring/sungather/config/config.yaml:/config/config.yaml
      - /volume2/docker/PRODUCTION/energy-monitoring/sungather/logs:/logs
    environment:
      TZ: "Europe/Berlin"
```

My config.yaml:
````yaml
inverter:
  host: 192.168.178.43                       # [Required] IP Address of the Inverter or Dongle
  port: 8082                               # [Optional] Default for modbus is 502, for http is 8082
  # timeout: 10                             # [Optional] Default is 10, how long to wait for a connection
  # retries: 3                              # [Optional] Default is 3, how many times to retry if connection fails
  # slave: 0x01                             # [Optional] Default is 0x01
  # scan_interval: 30                       # [Optional] Default is 30
  connection: http                     # [Required] options: modbus, sungrow, http
  model: "SH8.0RT"                        # [Optional] This is autodetected on startup, only needed if detection issues or for testing
                                            # See model list here: https://github.com/bohdan-s/SunGather#supported
  # serial: xxxxxxxxxx                      # [Optional] This is autodetected on startup, only needed if detection issues or for testing, used as a unique ID
  # smart_meter: True                       # [Optional] Default is False, Set to true if inverter supports reading grind / house consumption
  # use_local_time: False                   # [Optional] Default False, Uses Inventer time, if true it uses PC time when updating timestamps (e.g. PVOutput)
  # log_console: INFO                       # [Optional] Default is WARNING, Options: DEBUG, INFO, WARNING, ERROR
  # log_file: DEBUG                         # [Optional] Default is OFF, Options: OFF, DEBUG, INFO, WARNING, ERROR
  # level: 1                                # [Optional] Set the amount of information to gather
                                            # 0 = Model and Solar Generation, 
                                            # 1 (default) = Useful data, all required for exports, 
                                            # 2 everything your Inverter supports, 
                                            # 3 Everything from every register 

# If you do not want to use a export, you can either remove the whole configuration block
# or set enabled: False
exports:
  # Print Registers to console, good for debugging / troubleshooting
  - name: console         
    enabled: True                          # [Optional] Default is False

  # Runs a simple Webserver showing Config and last read registers
  # Access at http://localhost:8080 or http://[serverip]:8080
  - name: webserver 
    enabled: True                           # [Optional] Default is False
    # port: 8080                            # [Optional] Default is 8080

  # Output data to InfluxDB
  - name: influxdb
    enabled: True                          # [Optional] Default is False
    url: "http://192.168.178.5:18086"            # [Optional] Default URL is http://localhost:8086
    token: <redacted>                           # [Required] API Token OR Username:Password (for influxdb v1.8x comment it out and use username + password)
    # username:                             # [Optional] Username if not using token
    # password:                             # [Optional] Password if not using token
    org: "Default"                          # [Required] InfluxDB Organization (for influxdb v1.8x this will be ignored)
    bucket: "SunGather"                     # [Required] InfluxDB Bucket (for influxdb v1.8x this is the database name)
    measurements:                           # [Required] Registers to publish to bucket
      - point: "power"
        register: daily_power_yields
      - point: "power"
        register: export_to_grid
      - point: "power"
        register: import_from_grid
      - point: "temp"
        register: internal_temperature
      - point: "power"
        register: daily_pv_export
      - point: "power"
        register: load_power_hybrid
      - point: "power"
        register: battery_level
      - point: "power"
        register: total_active_power
      - point: "power"
        register: export_to_grid
      - point: "power"
        register: import_from_grid
      - point: "power"
        register: daily_direct_energy_consumption
      - point: "power"
        register: daily_battery_charge_from_pv
      - point: "status"
        register: run_state
      - point: "status"
        register: last_reset
      - point: "power"
        register: total_power_yields
      - point: "power"
        register: phase_a_voltage
      - point: "status"
        register: daily_running_time
      - point: "power"
        register: pv_power_of_today
      - point: "power"
        register: daily_pv_energy_yields
      - point: "power"
        register: direct_power_consumption_today_pv
      - point: "power"
        register: direct_power_consumption_pv
      - point: "power"
        register: export_power_from_pv_today
      - point: "power"
        register: export_power_from_pv
      - point: "power"
        register: battery_charge_power_from_pv_today
      - point: "power"
        register: battery_charge_power_from_pv
      - point: "power"
        register: total_pv_generation
      - point: "power"
        register: total_pv_export
      - point: "power"
        register: export_power_hybrid
      - point: "power"
        register: total_battery_charge_from_pv
      - point: "power"
        register: total_direct_energy_consumption
      - point: "power"
        register: battery_power
      - point: "power"
        register: self_consumption_of_day
      - point: "power"
        register: daily_import_energy
      - point: "power"
        register: total_import_energy
      - point: "power"
        register: daily_export_energy
      - point: "status"
        register: start_stop
      - point: "status"
        register: timestamp
      - point: "power"
        register: daily_import_from_grid
      - point: "power"
        register: daily_export_to_grid
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use docker-compose.yml
2. Use config.yaml
3. Start Container
4. Wait for some time
5. See logs

**Logfiles**
See '[production-energy-monitoring-sungather-1_logs.txt](https://github.com/bohdan-s/SunGather/files/13195022/production-energy-monitoring-sungather-1_logs.txt)'

**Environment (please complete the following information):**
 - Synology-NAS with DSM 7.2-64570 Update 3
 - Docker Version 20.10.23, build 876946a
 - Portainer Version 2.19.1

julianjwong, 2024-02-15:
Dumb noob question from  me but how can I merge this change into my own instance? How can I find the file to replace on HA? I found it in your branch but want to manually merge into my deployment.  Sorry I'm not a developer but have a basic understanding of deployment

Cuupa, 2024-05-11:
@julianjwong: 
1. Easiest Way: bohdan-s has to merge this pull-request and build a new version of the docker image.
2. Someone other has to take this pull request (maybe create a fork) and create a docker image by himself.
3. You create a docker container from the python-image and mount the source code as a volume.

Regarding 2.): I guess @vetinari, myself or you can create a new docker image, but the problem is, it would create unwanted fragmentation when pushed to a public registry. 

Like we create a image with something like vetinari/sungather or cuupa/sungather including this fix, but you'll need to switch to these images in your docker-compose or whatever you're using. 

When @bohdan-s is now releasing a new version, you'll need to switch back to the official docker-image. Might be a minor inconvenience for you, but if someone else did not get the memo, he/she'll maybe wonder why their stack get no updates etc (maybe a "synthetic" case, just trying to explain my thought process :D)

bohdan-s, 2024-05-11:
Hi. I'll try and merge and build this weekend :)

vetinari, 2024-05-11:
@Cuupa I went a different way and rewrote the whole thing in go, fixed and changed a few things there :) well, the hold registers are missing, currently I have no use for those (not yet(?) publicly available).

dmesser, 2025-04-14:
@bohdan-s Any chance we could merge this change? It's happening consistently for me with InfluxDB as the export target

bohdan-s, 2025-04-15:
Hi,
The root cause is the inverter isn't providing the registers with the data/tome. 

Using the config option:
use_local_time: True

In the config.yaml will actually fix the time not been collected by using the host time instead. 

I'll have a look at the change shortly.