bohdan-s/SunGather #144: SH8.0RT-V112: Include registers to Level 2

pejobo, 2023-11-06:
Hi,
I own an SH8.0RT and the api works fine, but the registers I'm interested in are only available on level 3:

```
$ ../bin/python3 sungather.py --runonce -v 40 2> /dev/null | grep  "total_active_power\|load_power_hybrid\|export_power_hybrid" 
| 5031    | total_active_power                  | 1159 W               |
| 13008   | load_power_hybrid                   | 144 W                |
| 13010   | export_power_hybrid                 | 1015 W               |
```

Is this possible to include theses on level 2?

If someone would give me a code pointer I could also try to create a PR for this.

closed 2023-11-06 by pejobo

dmindfuse, 2024-05-10:
@pejobo , has your issue been resolved? I can't get it to work with SH8.0RT either. And because I can't access the console in Portainer -issues with docker and ubuntu LTS- I'm unable to try the API. 

`2024-05-10 14:55:23 ERROR    MQTT: Configured to use state_battery_charging but not configured to scrape this register
MQTT: Configured to use state_battery_charging but not configured to scrape this register
2024-05-10 14:55:38 ERROR    index out of range
index out of range
2024-05-10 14:55:38 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):`

The logs go on, I get no data back. I tried levels 0-3 to no avail. Scratching my head.... Anyone? Thanks.

pejobo, 2024-05-24:
@dmindfuse as far as I remember I did just what I described above: manually configured my model in the config file. For this no console access inside a docker container should be necessary.

dmindfuse, 2024-05-25:
Hi @pejobo , yep, as you said. Forcing it worked. My sungrow reports SH8.0-V112, which isn’t listed as such. But it works almost perfectly trimming it SH8.0. My only remaining issue was to figure out why on mod is connection the charging/discharging was always negative. But I gave up on it, sadly, just using HTTP in the config now and that works fine.