bohdan-s/SunGather #146: FBC - Forced Battery Charge registers

CaHaGa, 2023-11-08:
Hi all,

working on my SH8.0RT in the SunGrow-Web-Config I determined the following holding registers.
Please cross-check and if found good, add to the registers-sungrow.yaml.
I didn't find them in any official doc.
Regards - CaHaGa

    - name: "forced_batt_charge"
      level: 2
      address: 13140
      datatype: "U16"
      models: ["SH8.0RT"]
      datarange:
        - response: 0xAA
          value: "Enable"
        - response: 0x55
          value: "Disable"
    - name: "forced_batt_charge_day"
      level: 2
      address: 13141
      datatype: "U16"
      models: ["SH8.0RT"]
      datarange:
        - response: 1
          value: "AllDaysOfWeek"
        - response: 0
          value: "WorkingDaysOnly"
    - name: "forced_batt_charge_1_start_hour"
      level: 0
      address: 13142
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_batt_charge_1_start_minute"
      level: 0
      address: 13143
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_batt_charge_1_end_hour"
      level: 0
      address: 13144
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_batt_charge_1_end_minute"
      level: 0
      address: 13145
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_batt_charge_1_SOC_perc"
      level: 2
      address: 13146
      datatype: "U16"
      models: ["SH8.0RT"]
      accuracy: 1
      unit: "%"
    - name: "forced_batt_charge_2_start_hour"
      level: 0
      address: 13147
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_batt_charge_2_start_minute"
      level: 0
      address: 13148
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_batt_charge_2_end_hour"
      level: 0
      address: 13149
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_batt_charge_2_end_minute"
      level: 0
      address: 13150
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_batt_charge_2_SOC_perc"
      level: 2
      address: 13151
      datatype: "U16"
      models: ["SH8.0RT"]
      accuracy: 1
      unit: "%"

CaHaGa, 2023-11-08:
Dang! After all that (fun) work I just found some other doc stating different registers [here](https://www.photovoltaikforum.com/thread/166134-daten-lesen-vom-sungrow-wechselrichtern-modbus/?postID=3452929#post3452929).
Since this one looks like an official one, use this ...

    - name: "forced_charge"
      level: 2
      address: 33208
      datatype: "U16"
      models: ["SH8.0RT"]
      datarange:
        - response: 0xAA
          value: "Enable"
        - response: 0x55
          value: "Disable"
    - name: "forced_charge_day"
      level: 2
      address: 33209
      datatype: "U16"
      models: ["SH8.0RT"]
      datarange:
        - response: 1
          value: "AllDaysOfWeek"
        - response: 0
          value: "WorkingDaysOnly"
    - name: "forced_charge_1_start_hour"
      level: 0
      address: 33210
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_charge_1_start_minute"
      level: 0
      address: 33211
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_charge_1_end_hour"
      level: 2
      address: 33212
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_charge_1_end_minute"
      level: 2
      address: 33213
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_charge_1_SOC"
      level: 2
      address: 33214
      datatype: "U16"
      models: ["SH8.0RT"]
      accuracy: 1
      unit: "%"
    - name: "forced_charge_2_start_hour"
      level: 2
      address: 33215
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_charge_2_start_minute"
      level: 2
      address: 33216
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_charge_2_end_hour"
      level: 2
      address: 33217
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "HH"
    - name: "forced_charge_2_end_minute"
      level: 2
      address: 33218
      datatype: "U16"
      models: ["SH8.0RT"]
      unit: "MM"
    - name: "forced_charge_2_SOC"
      level: 2
      address: 33219
      datatype: "U16"
      models: ["SH8.0RT"]
      accuracy: 1
      unit: "%"
    - name: "Load rated power"
      level: 2
      address: 33274
      datatype: "U16"
      models: ["SH8.0RT"]
      accuracy: 0.01
      unit: "kW"

michbeck100, 2023-11-09:
Thanks for your work. We have to discuss if this change can be made because there is still a major change in SunGather coming.