bohdan-s/SunGather #147: total_active_power only available in Level 3

picto3000, 2023-11-09:
My SG8.0RS requires Level 3 to provide total_active_power, which of course throws nonstop errors. Can I please request an update to allow me to drop down to Level 2. I’m using this via docker on Latest tag, which I’ve noticed may not be using the most up to date registers-sungrow.yaml. I've see other posts using v.0.2.3, and the file on guthub shows total_active_power is supported by SG8.0RS.

```
Starting SunGather 0.5.0
Help? https://github.com/bohdan-s/SunGather
NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
Loaded config: /config/config.yaml
Loaded registers: registers-sungrow.yaml
Registers file version: 0.2.2
```

```
  sungather:
    container_name: sungather
    image: bohdans/sungather:latest
    network_mode: host
    volumes:
      - ${USERDIR}/docker/sungather:/config
      - ${LOG}/sungather:/sungather/log
    << : *env
```

LEVEL 2
```
 +----------------------------------------------+
 | Inverter Configuration Settings              |
 +----------------------------------------------+
 | Config             | Value                   |
 +--------------------+-------------------------+
 | host               | 192.168.1.90            |
 | port               | 502                     |
 | timeout            | 10                      |
 | retries            | 3                       |
 | RetryOnEmpty       | False                   |
 | model              | SG8.0RS                 |
 | serial_number      | A236240987              |
 | level              | 2                       |
 | scan_interval      | 30                      |
 | use_local_time     | False                   |
 | smart_meter        | True                    |
 | connection         | modbus                  |
 | slave              | 1                       |
 | start_time         |                         |
 +----------------------------------------------+
 +----------------------------------------------------------------------+
 | Address | Register                            | Value                |
 +---------+-------------------------------------+----------------------+
 | ----    | device_type_code                    | SG8.0RS              |
 | vr001   | run_state                           | ON                   |
 | vr003   | last_reset                          | 2023-11-9 10:51:32   |
 | vr006   | daily_export_to_grid                | 0.0 kWh              |
 | vr007   | daily_import_from_grid              | 0.00013333333333333334 kWh |
 | 5001    | nominal_active_power                | 8.0 kW               |
 | 5002    | output_type                         | 2P                   |
 | 5003    | daily_power_yields                  | 3.9 kWh              |
 | 5004    | total_power_yields                  | 27 kWh               |
 | 5008    | internal_temperature                | 32.0 °C              |
 | 5011    | mppt_1_voltage                      | 287.6 V              |
 | 5012    | mppt_1_current                      | 2.8 A                |
 | 5017    | total_dc_power                      | 1684 W               |
 | 5019    | phase_a_voltage                     | 245.0 V              |
 | 5020    | phase_b_voltage                     | 0.0 V                |
 | 5021    | phase_c_voltage                     | 0.0 V                |
 | 5033    | total_reactive_power                | 1255 Var             |
 | 5035    | power_factor                        | 0.76                 |
 | 5036    | grid_frequency                      | 50.0 Hz              |
 | 5083    | meter_power                         | 16 W                 |
 | 5091    | load_power                          | 0 W                  |
 | 5113    | daily_running_time                  | 200 min              |
 | 5006    | start_stop                          | Start                |
 | 5007    | power_limitation_switch             | Enable               |
 | 5008    | power_limitation_setting            | 100.0 %              |
 | 5019    | power_factor_setting                | 0.0                  |
 | 5036    | reactive_power_adjustment_mode      | Enable Q(U)          |
 | 5037    | reactive_power_percentage_setting   | 0.0 %                |
 | 5039    | power_limitation_adjustment         | 8.0 kW               |
 | 5040    | reactive_power_adjustment           | 0.0 kVar             |
 | vr002   | timestamp                           | 2023-11-9 10:51:32   |
 | vr004   | export_to_grid                      | 0 W                  |
 | vr005   | import_from_grid                    | 16 W                 |
 +----------------------------------------------------------------------+
```

LEVEL 3

` | 5031    | total_active_power                  | 1703 W               |`

James-YL, 2023-11-28:
Please add for SG10RS as well, I’m using home assistance add-on. Thanks

susjus, 2024-01-23:
+1 Have a SG8.0RS using docker image on NAS - Please update the latest docker image to v0.5.1