bohdan-s/SunGather #150: External Meter data not showing

yoursnack, 2023-11-14:
I have a Sungrow SG40CX-V112 and I am trying to scrape it's data. For now I have data pertaining to the meter it self. On level 3 I get this:

+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| 5000    | device_type_code                    | SG40CX               |
| vr001   | run_state                           | ON                   |
| vr003   | last_reset                          | 2023-11-14 10:48:58  |
| vr006   | daily_export_to_grid                | 0.0 kWh              |
| vr007   | daily_import_from_grid              | 0.0 kWh              |
| 4950    | protocol_number                     | 0                    |
| 4952    | protocol_version                    | 0                    |
| 4954    | arm_software_version                | 0                    |
| 4969    | dsp_software_version                | 0                    |
| 4990    | serial_number                       | A216240191           |
| 5001    | nominal_active_power                | 40.0 kW              |
| 5002    | output_type                         | 3P4L                 |
| 5003    | daily_power_yields                  | 10.9 kWh             |
| 5004    | total_power_yields                  | 109535.4 kWh         |
| 5006    | total_running_time                  | 7756 h               |
| 5008    | internal_temperature                | 31.1 °C              |
| 5009    | total_apparent_power                | 3464 VA              |
| 5011    | mppt_1_voltage                      | 678.9 V              |
| 5012    | mppt_1_current                      | 0.7 A                |
| 5013    | mppt_2_voltage                      | 674.8 V              |
| 5014    | mppt_2_current                      | 0.8 A                |
| 5015    | mppt_3_voltage                      | 674.5 V              |
| 5016    | mppt_3_current                      | 1.9 A                |
| 5017    | total_dc_power                      | 3391 W               |
| 5019    | phase_a_voltage                     | 235.2 V              |
| 5020    | phase_b_voltage                     | 233.7 V              |
| 5021    | phase_c_voltage                     | 234.2 V              |
| 5022    | phase_a_current                     | 5.3 A                |
| 5023    | phase_b_current                     | 5.3 A                |
| 5024    | phase_c_current                     | 5.3 A                |
| 5031    | total_active_power                  | 3340 W               |
| 5033    | total_reactive_power                | 568 Var              |
| 5035    | power_factor                        | 0.98                 |
| 5036    | grid_frequency                      | 50.0 Hz              |
| 5038    | work_state_1                        | Run                  |
| 5040    | alarm_time_month                    | 0                    |
| 5041    | alarm_time_day                      | 0                    |
| 5042    | alarm_time_hour                     | 0                    |
| 5043    | alarm_time_minute                   | 0                    |
| 5045    | alarm_time_second                   | 0                    |
| 5045    | alarm_code_1                        | 0                    |
| 5049    | nominal_reactive_power              | 26.4 kVar            |
| 5071    | array_insulation_resistance         | 1174 k-ohm           |
| 5077    | active_power_regulation_setpoint    | 44000 W              |
| 5079    | reactive_power_regulation_setpoint  | 0 Var                |
| 5081    | work_state_2                        | Run                  |
| 5083    | meter_power                         | 0 W                  |
| 5085    | meter_a_phase_power                 | 0 W                  |
| 5087    | meter_b_phase_power                 | 0 W                  |
| 5089    | meter_c_phase_power                 | 0 W                  |
| 5091    | load_power                          | 3340 W               |
| 5093    | daily_export_energy                 | 0.0 kWh              |
| 5095    | total_export_energy                 | 0.0 kWh              |
| 5097    | daily_import_energy                 | 0.0 kWh              |
| 5099    | total_import_energy                 | 0.0 kWh              |
| 5101    | daily_direct_energy_consumption     | 0.0 kWh              |
| 5103    | total_direct_energy_consumption     | 0.0 kWh              |
| 5113    | daily_running_time                  | 198 min              |
| 5115    | mppt_4_voltage                      | 679.5 V              |
| 5116    | mppt_4_current                      | 1.7 A                |
| 5117    | mppt_5_voltage                      | 0.0 V                |
| 5118    | mppt_5_current                      | 0.0 A                |
| 5119    | mppt_6_voltage                      | 0.0 V                |
| 5120    | mppt_6_current                      | 0.0 A                |
| 5121    | mppt_7_voltage                      | 0.0 V                |
[... 57 lines cut ...]
| 5019    | power_factor_setting                | 0.0                  |
| 5035    | night_svg_switch                    | Disable              |
| 5036    | reactive_power_adjustment_mode      | Reactive power percentage setting  |
| 5037    | reactive_power_percentage_setting   | 0.0 %                |
| 5039    | power_limitation_adjustment         | 44.0 kW              |
| 5040    | reactive_power_adjustment           | 0.0 kVar             |
| 5041    | pid_recovery                        | Disable              |
| 5042    | anti_pid                            | None                 |
| 5043    | fullday_pid_suppression             | None                 |
| vr002   | timestamp                           | 2023-11-14 10:48:58  |
| vr004   | export_to_grid                      | 0 W                  |
| vr005   | import_from_grid                    | 0 W                  |
+----------------------------------------------------------------------+

What's missing here is the important things for me, import to grid and export. I believe they have added an extra meter to measure this as evidenced by the following picture:

[image]

and 

[image]

What is defined here as meter 1, that's the data I am lacking here. How can I read from this?

Thanks in advance!

bohdan-s, 2023-11-14:
Set smart_meter = true in the config?

yoursnack, 2023-11-14:
smart_meter: True is already set

yoursnack, 2023-11-15:
Any way I can try to scrape the data from modbus? Like try all combinations? If so, could you point me in the right direction how to do that?