bohdan-s/SunGather #153: Controlling sungrow through EMS registers

hacor, 2023-11-24:
Hi,

I was wondering if someone out here could perhaps help me out with my Sungrow inverter experiment. I have a SH10.0RT hybrid inverter with a Sungrow battery of 12.8kWh and I want to permanently control the charging/discharging power of the battery. Why:
- Avoid discharge to my EV
- Charge at cheap prices when no solar predicted
- Avoid high house peaks.

My software is running perfectly on Goodwe inverters, but Sungrow gives me strange behaviours. All settings are based on the modbus PDF 12/01/2023

What do I set:
- **EMS mode** _13050_ => 3 (external EMS mode)
- **Heartbeat** _13080_=> 20 (repeated every 9 seconds to keep EMS mode 3 alive)
- **Charge/discharge** _13051_ => 170 (0xAA) = Charge
- **Charge/discharge power** _13052_ => 2000 (The SH10.0RT needs a power value not a percentage)

What happens in this situation:
The inverter starts charging at 2000W (read register 13022) for about 1 seconds, than the power drops to 0 and after 3 seconds it gives me 2000W again for 1 second. In the meantime no fields were written in the inverter.

Good to knows:
- This does not happen on discharging the battery (13051 set to 187). In this case power field 13022 is perfectly stable
- **Min Soc** _13059_ = 4%
- **Max soc** _13058_ = 95%
- All fields seem good
- Export power limit is set to the inverter maximum (10kW)
- When I set it back to defaults:

```json
{
    "ems_power": 0,
    "mode": 204,
    "ems_mode": 0
  }
```

The inverter starts back to solar charging in self-consumption mode, which works perfectly.

I really hope someone can help me out, because I'm really stuck here.

Thanks!
kind regards

Hacor

jbardu, 2024-05-24:
did you get any good answer to these issues?
I search a lot and there is so much stuff on reading data but very little on controlling such things. It seems like an obvious thing. Why is it so hard. Or you solved it?

hacor, 2024-05-24:
@jbardu It seemed to had something to do with the grid import feed. I had to allow the inverter to take power from the grid to charge. It was fighting itself in order to do so.

As a hint: try your commands with `mbpoll` this tool helped me get the inverter stable by trying multiple registers

closed 2024-05-24 by hacor