bohdan-s/SunGather #154: Getting spikes using SunGather feeding into Home Assistant Energy dashboard

insertactivate, 2023-12-08:
Getting spikes using SunGather feeding into Home Assistant Energy dashboard when the values suddenly drop to 0 and then up again (see e.g. 2:27 in attached snapshot) to previous value that then adds to the accumulated energy for a day which often doubles the total solar energy for a day. 

[image]

[image]

[image]

insertactivate, 2023-12-08:
<h3>SunGather v0.5.1</h3><p></p><h4>Need Help?<span> </span><href a="https://github.com/bohdan-s/SunGather">https://github.com/bohdan-s/SunGather</href></h4><p></p><h4>NEW HomeAssistant Add-on:<span> </span><href a="https://github.com/bohdan-s/hassio-repository">https://github.com/bohdan-s/SunGather</href></h4><p></p>
Address
--

---- | device_type_code | SG12RT
vr001 | run_state | ON
vr003 | last_reset | 2023-12-9 0:00:39
vr006 | daily_export_to_grid | 0.0 kWh
vr007 | daily_import_from_grid | 0.0 kWh
5003 | daily_power_yields | 0.0 kWh
5004 | total_power_yields | 13057.2 kWh
5006 | total_running_time | 4295 h
5008 | internal_temperature | 16.1 °C
5009 | total_apparent_power | 0 VA
5019 | phase_a_voltage | 0.0 V
5031 | total_active_power | 0 W
5038 | work_state_1 | Standby
5083 | meter_power | 0 W
5091 | load_power | 0 W
5113 | daily_running_time | 0 min
5128 | monthly_power_yields | 10.6 kWh
5006 | start_stop | Start
vr002 | timestamp | 2023-12-9 0:15:36
vr004 | export_to_grid | 0 W
vr005 | import_from_grid | 0 W

<p>Total 21 registers</p><p></p><table><tbody><tr><th>Configuration</th><th>Value</th></tr><tr><td>host</td><td>10.0.30.200</td></tr><tr><td>port</td><td>502</td></tr><tr><td>timeout</td><td>10</td></tr><tr><td>retries</td><td>3</td></tr><tr><td>RetryOnEmpty</td><td>False</td></tr><tr><td>model</td><td>SG12RT</td></tr><tr><td>serial_number</td><td>A22A207075</td></tr><tr><td>level</td><td>1</td></tr><tr><td>scan_interval</td><td>30</td></tr><tr><td>use_local_time</td><td>False</td></tr><tr><td>smart_meter</td><td>True</td></tr><tr><td>connection</td><td>modbus</td></tr><tr><td>slave</td><td>1</td></tr><tr><td style="color: rgb(0, 0, 0); font-family: Times; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">start_time</td>

insertactivate, 2023-12-10:
Adding one more screenshot which covers 3 days (starting a bit late the first day). Events when data goes to 0 and then back again to previous value which then adds to the total energy sum and makes it incorrect (doubles it): day 1 at 6:24 PM, day 2 at 2:27 PM and 6:00 PM, day 3 at 10:45 AM and 3:35 PM.

[image]

martynaskel, 2023-12-21:
I have exactly the same problem. Event tried to create a secondary "utility meter" sensor from the Sungater's daily generation sensor with a "Periodically resetting" setting on, but that didn't help.

insertactivate, 2023-12-21:
So far I have "solved" it with a filter in Home Assistant; https://community.home-assistant.io/t/issues-in-energy-dashboard-with-accumulated-solar-production-values/652460/9

stniemin, 2024-05-28:
Same issue here. Thanks for the workaround tip @insertactivate!