bohdan-s/SunGather #155: registers 13017 and 13018-9 not returned

monojk, 2023-12-10:
Fow my SG8.0RT with SBR096 battery registers 13017 (Daily direct energy consumption) and 13018-9 (Total direct energy consumption) are not returned  although they are defined in the registers-sungrow.yaml. 
Is there a good way to debug this? I get massive data when using the -v 10 flag although one thing could be related (-v 20 flag):
`INFO     Scraping: 1/19 registers failed to scrape`

bohdan-s, 2023-12-10:
Are you using Modbus? Try via Http method. 
what registers are listed on the webui?

monojk, 2023-12-11:
I am using http. Setting is level 3 (all registers). Here are the registers:
```
Level 3
sungather@bc3af94f287d:/usr/src/sungather$ python sungather.py
2023-12-09 10:13:59 INFO     Starting SunGather 0.5.0
2023-12-09 10:13:59 INFO     Need Help? https://github.com/bohdan-s/SunGather
2023-12-09 10:13:59 INFO     NEW HomeAssistant Add-on: https://github.com/bohdan-s/hassio-repository
2023-12-09 10:13:59 INFO     Loaded config: config.yaml
2023-12-09 10:14:00 INFO     Loaded registers: registers-sungrow.yaml
2023-12-09 10:14:00 INFO     Registers file version: 0.2.2
+----------------------------------------------+
| Inverter Configuration Settings              |
+----------------------------------------------+
| Config             | Value                   |
+--------------------+-------------------------+
| host               | 192.168.178.124         |
| port               | 8082                    |
| timeout            | 20                      |
| retries            | 3                       |
| RetryOnEmpty       | False                   |
| model              | SG8.0RT                 |
| serial_number      | A227295906              |
| level              | 3                       |
| scan_interval      | 60                      |
| use_local_time     | False                   |
| smart_meter        | False                   |
| connection         | http                    |
| slave              | 1                       |
| start_time         |                         |
+----------------------------------------------+
2023-12-09 10:14:04 ERROR    Webserver: Error: [Errno 98] Address already in use
Webserver: Error: [Errno 98] Address already in use
2023-12-09 10:14:11 WARNING  No data returned for read, 7012:25
No data returned for read, 7012:25
+----------------------------------------------------------------------+
| Address | Register                            | Value                |
+---------+-------------------------------------+----------------------+
| 5000    | device_type_code                    | SG8.0RT              |
| vr001   | run_state                           | ON                   |
| vr003   | last_reset                          | 2023-12-9 10:14:10   |
| vr006   | daily_export_to_grid                | 0.0 kWh              |
| vr007   | daily_import_from_grid              | 0.0 kWh              |
| 4950    | protocol_number                     | 1094856704           |
| 4952    | protocol_version                    | 16781568             |
| 4954    | arm_software_version                | 16722                |
| 4969    | dsp_software_version                | 19780                |
| 4990    | serial_number                       | A227295906           |
| 5001    | nominal_active_power                | 8.0 kW               |
| 5002    | output_type                         | 3P4L                 |
| 5003    | daily_power_yields                  | 0.3 kWh              |
| 5004    | total_power_yields                  | 0.0 kWh              |
| 5006    | total_running_time                  | 0 h                  |
| 5008    | internal_temperature                | 23.8 °C              |
| 5009    | total_apparent_power                | 330 VA               |
| 5011    | mppt_1_voltage                      | 421.7 V              |
| 5012    | mppt_1_current                      | 1.0 A                |
| 5013    | mppt_2_voltage                      | 343.9 V              |
| 5014    | mppt_2_current                      | 0.9 A                |
| 5015    | mppt_3_voltage                      | 0.0 V                |
| 5016    | mppt_3_current                      | 0.0 A                |
| 5017    | total_dc_power                      | 679 W                |
| 5019    | phase_a_voltage                     | 236.8 V              |
| 5020    | phase_b_voltage                     | 237.2 V              |
| 5021    | phase_c_voltage                     | 238.0 V              |
| 5022    | phase_a_current                     | 0.6 A                |
| 5023    | phase_b_current                     | 0.6 A                |
| 5024    | phase_c_current                     | 0.6 A                |
| 5031    | total_active_power                  | 328 W                |
| 5033    | total_reactive_power                | -6 Var               |
| 5035    | power_factor                        | 1.0                  |
[... 139 lines cut ...]
| 5011    | export_power_limitation_value       | 0                    |
| 5012    | current_transformer_output_current  | 0 A                  |
| 5013    | current_transformer_range           | 0 A                  |
| 5014    | current_transformer                 | None                 |
| 5015    | export_power_limitation_percentage  | 0.0 %                |
| 5016    | installed_pv_power                  | 0.0 KW               |
| 5019    | power_factor_setting                | 0.0                  |
| 5035    | night_svg_switch                    | None                 |
| 5036    | reactive_power_adjustment_mode      | Off                  |
| 5037    | reactive_power_percentage_setting   | 0.0 %                |
| 5039    | power_limitation_adjustment         | 0.0 kW               |
| 5040    | reactive_power_adjustment           | 0.0 kVar             |
| 5041    | pid_recovery                        | None                 |
| 5042    | anti_pid                            | None                 |
[... cut ...]
| 13085   | start_discharging_power             | 0 W                  |
| 13086   | energy_meter_comm                   | Enabled              |
| 13100   | soc_reserve                         | 0 %                  |
| vr002   | timestamp                           | 2023-12-9 10:14:10   |
| vr004   | export_to_grid                      | 0 W                  |
| vr005   | import_from_grid                    | 0 W                  |
+----------------------------------------------------------------------+
Logged 194 registers to Console
```

monojk, 2023-12-12:
A correction: inverter is SH8.0RT not SG8.0RT. Problem is the same.